commaai/openpilot #2316: Needs a better way to detect hotspot

cydia2020, 2020-10-11:
On my Pixel 4 XL running Android 11, the hotspot's IP address starts with 192.168. and ends with 143; the only variable in this ip address is the third part of it, breaking the hotspot detection function in uploader. openpilot should probably have a better way of detecting hotspots.

eg: list wi-fi name as a hotspot when the car is above 50 km/h and the said SSID is connected.

adeebshihadeh, 2020-10-12:
I’m not sure hotspot detection is useful anymore since we don’t upload raw logs while onroad. What’s your use case for this?

cydia2020, 2020-10-12:
Hi, sometimes I do a bit of tinkering in the car while I am connected to my hotspot. However this might be an isolated case and if it doesn't have widespread affect I will just make the changes in my fork and close the issue. Thank you.

closed 2020-10-12 by cydia2020