commaai/openpilot #2337: when i scons use_webcam=1

ScienceZhi, 2020-10-14:
**Describe the bug**
I cloned the latest code and tried to run openpilot. using ubuntu.
When I execute scons web_camera=1, I produce an err that can be easily solved twice.
The first is the deprecation of the clCreateCommandQueue function in opencl.
One is that there is no err variable.

**How to reproduce or log data**
selfdrive/camerad/main.cc:194:24: error: 'clCreateCommandQueue' is deprecated
      [-Werror,-Wdeprecated-declarations]
  cl_command_queue q = clCreateCommandQueue(s->context, s->device_id, 0, &err);
                       ^
/usr/include/CL/cl.h:1446:72: note: 'clCreateCommandQueue' has been explicitly
      marked deprecated here
  ...cl_int *                       /* errcode_ret */) CL_EXT_SUFFIX__VERSION...
                                                       ^
/usr/include/CL/cl_platform.h:119:74: note: expanded from macro
      'CL_EXT_SUFFIX__VERSION_1_2_DEPRECATED'
  ...#define CL_EXT_SUFFIX__VERSION_1_2_DEPRECATED __attribute__((deprecated))
                                                                  ^
1 error generated.

and

selfdrive/camerad/cameras/camera_webcam.cc:247:7: error: unused variable 'err'
      [-Werror,-Wunused-variable]
  int err;
      ^
selfdrive/camerad/cameras/camera_webcam.cc:265:3: error: use of undeclared
      identifier 'err'
  err = pthread_create(&rear_thread_handle, NULL,
  ^
selfdrive/camerad/cameras/camera_webcam.cc:267:10: error: use of undeclared
      identifier 'err'
  assert(err == 0);
         ^
selfdrive/camerad/cameras/camera_webcam.cc:271:3: error: use of undeclared
      identifier 'err'
  err = pthread_join(rear_thread_handle, NULL);
  ^
selfdrive/camerad/cameras/camera_webcam.cc:272:10: error: use of undeclared
      identifier 'err'
  assert(err == 0);
         ^
5 errors generated.

**Expected behavior**

**Additional context**

Operating system: [e.g. Ubuntu 16.04]

grekiki, 2020-10-14:
WIP https://github.com/commaai/openpilot/pull/2070

grekiki, 2020-10-14:
I think that pull request will solve the issue, so I'll close this.

SyazwanDev, 2023-05-11:
> 

Hi ScienceZhi, may I know which version of OpenPilot are you using?