commaai/openpilot #2358: Comma 2 will not engage

dwnaes, 2020-10-18:
Been using Comma 2 for months w no issue.  Openpilot will not engage

When trying to engage OP get the error msg, "Openpilot not available, issue with communication processes"

1>  Accelerate over 25mpg
2>  Attempt to engage OP
3>  OP will not engage.  Error Msg, '"Openpilot not available, issue with communication processes"
4>  Rinse/repeat

Expected behavior is OP engaging over 25mph.

 - Device: Comma 2
 - Dongle ID:  39eca5359e0b07f8
 - Route: 39eca5359e0b07f8|2020-10-18--17-34-33
 - Timestamp: immediately upon attempt to engage OP
 - Version: OP v0.7.9-Release
 - Car make/model: 2017 Toyota Corolla LE

Panda Dongle ID:  184abcf34150e3ea
don.naes@gmail.com
*We checked all hard wire connections, factory reset OP/installed...no luck :(

pd0wm, 2020-10-19:
Do you have a route that clearly shows the issue? Also make sure it's uploaded (connect your C2 to wifi), the linked route is not uploaded and is only 60 seconds long.

dwnaes, 2020-10-19:
Hi Willem,

Newbie here.  How do I create the needed log file?  I tried 'Download log segment(bz2) and 'View CAN Data in Cabana'(csv) neither of which I can attach to this ticket bc file type is not supported.  Next, went to the admin area and copy/pasted this to txt file
[comma2.txt](https://github.com/commaai/openpilot/files/5403012/comma2.txt)

Thank You.

[image]

pd0wm, 2020-10-19:
As you can see in that screenshot the 4th column still contains "None" instead of the filename for the video files. If everything is uploaded all columns should show filenames.

dwnaes, 2020-10-19:
Looking back thru the log files.  File uploads stopped working same day as OP.  I made this route 'public'

39eca5359e0b07f8|2020-10-18--18-52-05

May be the Communication error i'm getting when attempting to engage OP is the reason files are not uploading?

adeebshihadeh, 2020-10-22:
Looks like this is a hardware issue with your front camera. Contact support and we'll take care of it. Also adding an alert for this case that will show instead of comm issue.

closed 2020-10-22 by adeebshihadeh