commaai/openpilot #2361: Drives into Double yellow lines on a 2 lane highway

patcoghm, 2020-10-19:
Drives into double yellow lanes

2020 Prius Prime XLE, DSU unplugged, Dynamic cruise engaged. While driving on 2 lane US highways US95 in Nevada the car will move into the double yellow lines to the point of trying to straddle the lines.  Happens during darkness, 2 lanes with the no passing lines on the left.  BUT if I turn off cruise and turn on with a 1.5 second or longer it engages standard criuse without radar or long control with open pilot, just lateral control works.  During standard cruise it did not move into the yellow lines.  When in Dynamic cruise with full OP long and lateral  the car again moved into double yellow lines.

In this video "https://my.comma.ai/e4a16227cff292b5/1602985741199/1602992924316#"
time stamped Oct 17  200330 and Oct 17 201645

I expected it to not drive into the center of the road. Happens only with double yellow lines for no passing.

 - Device: Comma2
 - Dongle ID: e4a16227cff292b5
 - Route: https://my.comma.ai/e4a16227cff292b5/1602985741199/1602992924316
 - Timestamp: Oct 17 200330
                      Oct 17 201630
 - Version: open pilot V0.7.9-Release no forks
 - 2020 Toyota Prius Prime XLE
- Calibration 0.9 down, 0.1 left

This was a return trip from Phoenix, AZ to Reno.  The first leg was mentioned in openpilot -experience after the original trip to Phoenix with all the same problems happened.  Both happened about 8:00 PM PDT in total darkness on the Oct 12 just before Gold Hill NV on a freshly coated and lined section with double yellow lines.  The return trip for the bug report happened Oct 17 just past Mercury, NV leaving a divided highway to 2 lane US95. Others on Discord also have mentioned double yellow line problems in 2 lane US highways and driving into the lines.  Its a hard pull into the lines using opposite resistance on the steering wheel.
Original problem was in the drive "https://my.comma.ai/e4a16227cff292b5/1602556277800/1602561465865#" and was mentioned on 10/14/2020 on Discord openpilot-experience.

HaraldSchafer, 2020-10-20:
Thanks for the report, the neural network fails to understand how the car is moving because there are no features to track in the image. There are only continuous lines and nothing else of interest. This then messes up calibration and moves the car. We are working on improving the model in this situation.

xps-genesis, 2020-10-20:
@patcoghm Thank you for posting this bug. This has been an issue for a long time with multiple users discussing on discord. thank you for posting up the official bug report.
1>) night time
2>) double yellow middle lane line
3>) typically turning right however this may be just a coincidence and doesn't depend on the direction on the road curve.

@HaraldSchafer thank you for looking into it. It makes OP unusable on the same road during the return trip when dark.

HaraldSchafer, 2020-10-27:
https://github.com/commaai/openpilot/issues/2399

HaraldSchafer, 2020-10-27:
I cannot find any fully uploaded routes where this happens. If anyone experiences this again and could have the data upload over wifi and post the route here, that would be very helpful!

illumiN8i, 2020-10-29:
@HaraldSchafer I may have found a couple of examples:
46f99988a32768fc|2020-09-30--19-44-50--3
46f99988a32768fc|2020-09-30--19-44-50--6

DarkSlaayer, 2020-10-30:
> @patcoghm Thank you for posting this bug. This has been an issue for a long time with multiple users discussing on discord. thank you for posting up the official bug report.
> 1>) night time
> 2>) double yellow middle lane line
> 3>) typically turning left however this may be just a coincidence and doesn't depend on the direction on the road curve.
> 
> @HaraldSchafer thank you for looking into it. It makes OP unusable on the same road during the return trip when dark.

I experienced this to the T on my way home on a long road trip double yellow lines, only on curves to the left and night time never had it happen to me before but it seemed like it did it every time on this road was a new asphalt road and pitch black out. Trying to get the clip to load but it's a 6hr clip and no luck so far. This was in a 2020 Hyundai sonata btw.

HaraldSchafer, 2020-10-30:
@illumiN8i Thanks!

brittonx, 2020-10-31:
I can confirm this is a real issue.  It does seem to be worse at night where I've had it cross almost halfway into the opposite lane!

HaraldSchafer, 2020-10-31:
@brittonx more routes with this happening would be useful!

brittonx, 2020-11-01:
Here is the most recent segment...
2eacef693c0640f9|2020-10-30--18-44-42--12

SonyUSA, 2020-11-03:
Here you go, it's a huge mountain pass full of twisty turns at night and double center lane lines. Sometimes I let the car run over, but we got annoyed and started manually correcting it after a while. This was my return trip, I can share the outbound trip too if you need!

It's probably at least an hour long so lots of examples you can pull out:

2b979b9557651148|2020-10-26--01-37-40

EthernetIp, 2020-11-16:
5df6ac6dbf953ed1|2020-11-15--17-08-30--15
Practically not drivable at night when road has double yellow.

philbonk1977, 2020-11-16:
Here's my 2020 Prius Prime LE, left turn, same issue I think:
04befd90fd4f041c|2020-11-15--16-50-03--4

HaraldSchafer, 2020-11-30:
Another occurrence of this 
https://github.com/commaai/openpilot/issues/2652

HaraldSchafer, 2020-11-30:
I'm making progress on finding a fix for this problem.

csouers, 2020-12-06:
Here's another segment. This happened last night on master-ci from 2-3 days ago. (commit d1836caf32747a3c7206b7efb2b32fe7c65cdafe).

Segment: 2ad2b7955c2b304e|2020-12-05--22-20-24--6

I will also echo that this has been an issue for some time. Usually I have to disengage due to traffic but this time I was able to let it run its course. I have thought it was a fluke with this area, but it does it in a similar way each and every time (when dark outside). It does eventually correct itself, specifically after leading it back to the center of the lane after the bad curve in that segment.

HaraldSchafer, 2020-12-08:
This seems to be the underlying cause. We track features in the image to track the motion if the vehicle. However in scenes like the ones in the image it matches points on the laneline to random other points on the laneline. You can see correct matches on the sign on the right. These garbage matches cause the localizer to think it is not turning as much as it is. This messes up the ground truth for the model, which then makes the model predict lanelines too much into the turn in these situations.

[image]

I have a fix in the works!

HaraldSchafer, 2020-12-16:
Should be fixed once this PR is merged
https://github.com/commaai/openpilot/pull/19510

HaraldSchafer, 2020-12-16:
Here is the old model failing

[image]

and new model on same frame

[image]

HaraldSchafer, 2020-12-16:
feel free to open new issues if this still happens with new model

closed 2020-12-16 by HaraldSchafer