commaai/openpilot #2376: open pilot disengages when approching a stopped car at a light

gricha01, 2020-10-20:
**Describe the bug**
I have a 2019 Honda Passport. With open pilot engaged, 40mph on the city streets, if there is a car stopped at the red light in front of me as i approach it, it does not even try to slow down. i have to take over and initiate the stop. if i remember correctly it just shuts off, no beeboop.

**How to reproduce or log data**
you can reproduce it by driving toward a car at a stop light, does it every time.

**Expected behavior**
I am expecting it to start to slow down till it reaches its lowest ACC speed of 26 mph

**Device/Version information **
 - Device: comma two
 - Dongle ID: d5450ec7974784ca
 - Route: d5450ec7974784ca\|2020-10-20--17-06-17 
 - Timestamp: 17:09:15/17:10:09/17:11:22/17:13:12/17:16:12
 - Version: openpilot v0.7.9-release
 - Car make/model: Honda Passport 2019

**Additional context**
I went on a trip 2 days ago and this may be tied in. while on a 2 1/2 hour drive down the freeway it would pace cars no problem, but if a car changed lanes into my lane it would just disengage, no beeboop. just disengage.

gricha01, 2020-10-26:
So, any updates? What should I do?

pd0wm, 2020-11-09:
[image]

I looks like your car has a little offset in the USER_BRAKE signal. When openpilot sends a brake command this brings it just over the threshold for the brake detection, causing a disengage. Specifically this check triggers: https://github.com/commaai/openpilot/blob/master/selfdrive/car/honda/carstate.py#L309.

Can you try removing that check (which according to the comment is only for pedal users), and see if that fixes your issue?

gricha01, 2020-11-09:
I'm sorry, only had this for 20 days. how do i remove it? I've only had it plugged into the car so for. 
Tell me what to do, where to go in it. and what to remove and I'll do.  
Thanks George

gricha01, 2020-11-22:
Can you or someone there remove it? i don't have a clue how to remove it. I tried asking in discord, but got no help.

adeebshihadeh, 2021-06-17:
Is this still an issue?

gricha01, 2021-06-18:
Yes. I still have the issue. I do not know how to SSH. Is there someone you can recommend or who I can look for on discord to correct this? When I ask I only get told to go to YouTube and learn SSH. I'm not a computer person. I just would like to hire someone to tap into it and fix it.

sshane, 2021-06-21:
The offsets seem to be different across the three cars where it checks if `ret.brake` > 0.05. The Ridgeline's offset is 0.02 where it has no noise, and the Passport is 0.05 with lots of noise: 
[image]
[image]

Should we fix this in the DBCs, or remove this check altogether? From what I've seen `cp.vl["POWERTRAIN_DATA"]["BRAKE_PRESSED"]` is catching all of the brake press cases even at low speeds.

adeebshihadeh, 2021-06-21:
I checked the commit in one responsible for those lines. It seems to be a mitigation for a grinding noise when disengaging with the brake pedal at low speeds. Since then, we've merged a few PRs to address other grinding noises. Do we still need that code?

justinmccaffrey, 2021-11-06:
Are you still having this issue? I've never had this with my 2021 passport (Openpilot Sept 2021-Present)

gricha01, 2021-11-07:
Yes I am.

gricha01, 2021-11-07:
Yes I am.

fraserxiong, 2022-01-10:
Hi, Can I create a temporary branch that has a threshold of 0.6 without getting banned by comma?

adeebshihadeh, 2022-01-30:
This should be fixed by #23641

closed 2022-01-30 by adeebshihadeh

gricha01, 2022-02-23:
I am running V0.8.13 and can confirm this is no longer happening. The only thing i am getting now is the same that is listed in #23821. thanks everyone.