commaai/openpilot #2383: Unit test for Toyota safety param

adeebshihadeh, 2020-10-21:


ShaneSmiskol, 2020-10-22:
Here's my last 14 drives that had enough samples for a prediction, the current script doesn't seem very accurate. 

[image]

I think the varying results could be caused by random occurrences when EPS and CMD torque become out of sync, like the following examples:

[image]

And:

[image]

ShaneSmiskol, 2020-10-22:
If I filter the data in the script and only allow data where the difference between CMD and EPS is less than 100 it seems a bit better, could go lower with the threshold
[image]

```
Raw data:      mean: 0.9235, std.: 0.0524
Filtered data: mean: 0.9171, std.: 0.0389
```

ShaneSmiskol, 2020-11-29:
I think #2647 will fix these issues

adeebshihadeh, 2021-06-08:
Appears the factor is non-linear. We need to understand it better.

closed 2021-06-08 by adeebshihadeh