commaai/openpilot #2389: Fix frequent log corruption

adeebshihadeh, 2020-10-22:


pd0wm, 2020-11-11:
Loggerd has to be sigkilled sometimes. We have to figure out why it doesn't exit cleanly all the time.

adeebshihadeh, 2020-11-18:
@ZwX1616 I looked into a few of the instances of SIGKILLing loggerd, and all were waiting for one or both encoders to exit.

Here's one
```
Nov 18, 2020 @ 13:05:06.448 | killing loggerd with SIGKILL

Nov 18, 2020 @ 13:05:06.094 | slow frame(123581) time: 100.90

Nov 18, 2020 @ 13:05:05.961 | slow frame(123580) time: 81.23

Nov 18, 2020 @ 13:05:03.580 | slow frame(123520) time: 84.91

Nov 18, 2020 @ 13:05:03.463 | slow frame(123519) time: 118.86

Nov 18, 2020 @ 13:05:02.095 | slow frame(123485) time: 99.94

Nov 18, 2020 @ 13:05:01.575 | set state event 0x1

Nov 18, 2020 @ 13:05:01.573 | set state event 0x2

Nov 18, 2020 @ 13:05:01.563 | encoder alt destroy

Nov 18, 2020 @ 13:05:01.559 | set state event 0x1

Nov 18, 2020 @ 13:05:01.555 | set state event 0x2

Nov 18, 2020 @ 13:05:01.547 | encoder destroy

Nov 18, 2020 @ 13:05:01.448 | joining threads

Nov 18, 2020 @ 13:05:01.447 | killing loggerd
```

ZwX1616, 2021-03-24:
Looked into the sigkills, and it seems to be one of the visionstream servers locking up due to exiting race condition between camerad and loggerd. A lot has changed in the visionipc2 refactor, so this is no longer relevant.

In terms of the corrupted logs, they largely correlate well to the device being unplugged shortly after a drive.

closed 2021-03-24 by ZwX1616