commaai/openpilot #2390: Take Control warnings while OP not engaged & when stopped, but looking away

Kats3y3, 2020-10-22:
**Describe the bug**

#1 - Getting ‘Take Control’ warnings while driving, but OP is not engaged.  #2 - Also, get ‘Take Control’ warnings while stopped (i.e. at a stop light), but I’m looking away (i.e. face recognition timeout).

**How to reproduce or log data**

As described above.  #1 happens if I’m driving, OP is not engaged, but I drift over a line a small amount or enter a turn too fast.  #2 happens every time I’m stopped, with OP engaged, but I’m looking away.

**Expected behavior**

#1 - One would expect to not get ‘Take Control’ warnings if you do not have OP engaged.  
#2 - One would expect OP to know that you are stopped and that it is OK to look away and therefore not give a ‘Take Control’ warning.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID:  6ffdd0db49c2f5e0
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version:  0.7.9
 - Car make/model: 2018 Honda Accord Hybrid Touring

**Additional context**

adeebshihadeh, 2020-10-22:
Can you post specific instances (routes with timestamps of the events)? I looked through a couple routes from your device and didn't find anything.

VirtuallyChris, 2020-10-23:
Both of these sound normal. Driver monitoring shows the green alert if you are distracted while at a stop but won't beep at you, and the other one is the lane departure warning alert which has a toggle in settings.

closed 2020-10-23 by VirtuallyChris

Kats3y3, 2020-10-26:
@VirtuallyChris  @adeebshihadeh   

I will agree on the driver monitoring alert.  It doesn’t beep at you, but does have the alert when you are stopped.  One could argue either way whether that is necessary.  

The lane departure alert in settings turns off ALL alerts (I assume).  Mine is going off when OP is not engaged.  Again, one could argue this is a ‘feature’ and not a ‘bug’.  ;)  I would land on the side that thinks if I do not have OP engaged, I do not want alerts from it.  I would recommend the lane alerts be disabled when OP is not engaged or at least make it a toggle.  

I notice the lane departure alerts on today’s drives around 0508 and 1509.

Thanks for both of your help.

Kats3y3, 2020-10-30:
I don’t agree that this is ‘Closed’.  George Hotz specifically said in a video on YouTube that OP would not alert you if not engaged.  Will have to find the exact reference, but was in interview with Lex Fridman.  Please reopen and address #1 above.  Thanks.  @VirtuallyChris @adeebshihadeh

adeebshihadeh, 2020-10-30:
By default, lane departure warnings are enabled and will alert you while not engaged, but they can be disabled in the settings.

Kats3y3, 2020-10-30:
https://youtu.be/-qf1NzI1I7E

“If the car is not moving, we do not actively enforce driver monitoring.”  George Hotz at 0:52

And again at 1:52.

@adeebshihadeh  When you say it can be disabled in settings, does that just disable it when OP is not engaged or all the time?  I’ll admit that I didn’t try to disable the setting because I don’t want to disable them all the time.  Just when OP is not engaged.  (Which seems logical.....at least to me.). Thanks!

Kats3y3, 2020-10-30:
To clarify, in video, George is really talking mostly about #2.  I’m OK with the OP ‘silent’ alerts if I’m stopped.  My issue is still with #1.  I.e. getting alerts from OP when it is not engaged.  I think George would agree with me.  ;)