commaai/openpilot #2393: 0.7.9 qt_sound cuts off audio early

SonyUSA, 2020-10-23:
**Describe the bug**

Audio alerts only play a few milliseconds/1 second before cutting off

**How to reproduce or log data**

Play back any audio alert (mono, PCM-16, .wav format) longer than ~1 second and it will cut off before the end

**Expected behavior**

Entire audio file should play before stopping

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Version: devel 0.7.9 
 - Car make/model: Honda Civic Hatch

**Additional context**

Looks like this may be an issue in regards to the new qt_sound introduced in 0.7.9 as all audio worked in prior versions and played until the end of the file.

SonyUSA, 2020-10-23:
Never mind, I see that audio times are hardcoded instead of scanning for audio file length, e.g. :

Priority.LOW, VisualAlert.none, AudibleAlert.none, .2, .2, .2),

What about generating a file with audio attributes at compile time if the concern is CPU usage to scan the file on playback?

SonyUSA, 2020-10-23:
Looking back, here is a snippet of a custom event I had made:

`
  EventName.promptDriverDistracted: {
    ET.WARNING: Alert(
      "PAY ATTENTION DINGUS",
      "Driver Appears Distracted",
      AlertStatus.userPrompt, AlertSize.mid,
      Priority.MID, VisualAlert.steerRequired, AudibleAlert.chimeDriver, .1, .1, .1),
  },
`

Which seems to indicate that it was, previously, playing until file end before. I'm gonna re-open the issue just to get confirmation...

adeebshihadeh, 2020-10-23:
The sound code hasn't changed for comma two in 0.7.9, but it did move to selfdrive/ui/android/sl_sound.cc. Does this issue happen with the stock sounds?

SonyUSA, 2020-10-24:
Just looking at selfdrive/ui/android/sl_sound.cc --

setVolume(volume);

Was removed from 

bool Sound::init(int volume) {

in 0.7.9 and the references from Sound were changed to SLSound. I'll dig around a bit and see if there were any other changes to the audio code and try out some different sound files. The sound mappings were moved as well to another file but I don't think that's an issue.

SonyUSA, 2020-10-29:
Haven't had time to swap out code, just took two 10 hour drives, I did notice that when activating OP and clicking the speed up button rapidly it will play the full audio clip, engaging it from a stop or just after the C2 is powered on, it will play no sounds at all. This may be related to the speed adjusted volume settings, so the removal of setVolume may actually be the culprit. I'll test when I can.

adeebshihadeh, 2020-11-10:
Closing this due to inactivity. Reopen if you can reproduce with stock openpilot.

closed 2020-11-10 by adeebshihadeh