commaai/openpilot #2395: HKG resume logic not working on Hyundai Genesis(2015-2016)

xps-genesis, 2020-10-24:
**Describe the bug**

The new resume logic in Master-Ci for HKG doesn't work on the Hyundai Genesis(2015-2016), This can be an issue with other older HKG cars. The resume logic is triggered when the standstill flag from SCC is set. This often allows the SCC to accept once or twice during a stop but often misses when the lead starts to move. 

**How to reproduce or log data**

ACC activated behind a stopped target vehicle to start the resume logic 

**Expected behavior**

Resume and continue moving when the lead starts to move.

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [6fe86b4e410e4c37]
 - Route: [6fe86b4e410e4c37|2020-10-23--19-14-59]
 - Timestamp: [497 seconds]
 - Version: [master-ci]
 - Car make/model: [Hyundai Genesis 2015-16]

adeebshihadeh, 2021-04-07:
Is this still an issue?

xps-genesis, 2021-04-11:
yup still doesn't work on older vehicles

SeanCox, 2023-02-03:
I tested the PR and left some notes for this https://github.com/commaai/openpilot/pull/25579

SeanCox, 2023-03-10:
@adeebshihadeh This issue has been fixed.

closed 2023-03-10 by adeebshihadeh