commaai/openpilot #2441: Check Forward Collision

haddsomnus, 2020-10-28:
When the car starts, I'm getting this error message "Check Forward Collision Avoidance Assist system." The openpilot is working fine, but seeing this error message over and over again is annoying. 

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [6b5520e0ab1c7118]
 - Route: [6b5520e0ab1c7118|2020-10-23--11-28-33--0]
 - Timestamp: [start up of the car]
 - Version: [release2]
 - Car make/model: [Santa fe 2019]

pd0wm, 2020-11-09:
Does the message show only once on startup and then goes away? Or is the message persistent?

adeebshihadeh, 2020-11-09:
Maybe this is fixed by #2490?

haddsomnus, 2020-11-09:
Yes, when I unplug the c2 the message disappears. I will try #2490 tonight

haddsomnus, 2020-11-11:
Update, #2490 did not work the error message still present.

adeebshihadeh, 2020-11-25:
If you disable openpilot via the settings (must be done before starting the car), does the message appear? Can you get a route with the stock system?

haddsomnus, 2020-11-27:
Yes, the message still appears, and even I unplug the c2. The message only disappears when I plug back the stock harness. 
Route: 6b5520e0ab1c7118|2020-11-26--17-24-50

pd0wm, 2020-11-27:
Can you post a clear picture of your harness with all the wires and the connectors visible? Maybe a wire is missing?

haddsomnus, 2020-11-28:
https://imgur.com/a/kvfZkds

vanillagorillaa, 2020-11-28:
@haddsomnus Are you located in the US? Can you please also post pictures of the vehicle harness itself?

vanillagorillaa, 2020-11-28:
@haddsomnus I believe I see your issue. If you compare your harness against this wiring diagram, https://github.com/commaai/neo/blob/master/car_harness/Type_II_Hyundai_Car_Harness.sch.pdf, you will see that on the tan 26 pin connector, Pins 8/10 (the blue and pink wires) are flipped. You can use a set of tweezers to lift the retention mechanism on the connector so that you can swap the wires, or reach out to support@comma.ai to have a new one swapped for you.

haddsomnus, 2020-11-28:
@vanillagorillaa  Thank you, I will contact comma support immediately.

closed 2020-11-28 by adeebshihadeh