commaai/openpilot #2446: Roads in EU

nickn17, 2020-10-29:
Roads in EU are narower than US, but now it's lane width hardcored to 3.7m  (secondary/tertiary road lane width is in Slovakia 3 meters)  
https://github.com/commaai/openpilot/blob/master/selfdrive/controls/lib/lane_planner.py
    self.lane_width = 3.7  -> 3.0

And next problem on secondary/tertiary with middle line only. OP is not often usable there, because car is going with right wheels on road edge
 speed_lane_width = interp(v_ego, [0., 31.], [2.8, **3.5**])  ->   speed_lane_width = interp(v_ego, [0., 31.], [2.8, **2.9**])
-> It's possible to add item to menu (EU roads, or configurable lane width?), opparams

HaraldSchafer, 2020-10-30:
We want to reduce reliance on lanelines and have the model control where to drive more so this is no longer an issue. We do no intend to change much to this logic before then.

closed 2020-10-30 by HaraldSchafer

cooldemo, 2020-10-30:
I am having the same issue with narrow roads in Europe and I believe this issue is Europe-wide. For this single reason I am forced to maintain a fork of OP. When is the model change planned to happen ?
My opinion is that customization is better, It seems you are trying to make "one driving option to suite them all". I like driving more towards the right lane, but there is no option for that. Some people like driving more to the center. I would like to have the option to switch to narrower roads by default without playing with the code.

HaraldSchafer, 2020-10-30:
Model improvements are a priority for us, it is unclear when it will be good enough so narrow lanes will work as well as wide lanes, but we are making progress.

We want openpilot to drive well in as many situations as possible out of the box without customization. If you want a lot of customization, that is a good reason to have a fork.