commaai/openpilot #2453: Make the time from initial wheel touch alert to making noise longer

ShaneSmiskol, 2020-10-30:
I just took my first drive with 0.7.10 (since 0.7.7) tonight and since I don't have DM at night I use wheel touch. However with the changes to wheel touch timings in this commit https://github.com/commaai/openpilot/commit/21d4ac1eebba06769f03edf7220ce84bd6ae154d I find myself not being able to catch the alert fast enough so that it *doesn't* make noise. If I'm looking straight at the road, as soon as I notice the alert to touch the wheel comes up, it just turns orange and starts making noise, possibly waking up anyone else in the car (but is also annoying if it's just you).
Is there any way we can either delay this prompt noise, or revert part of that commit to make the time from initial "touch the wheel" alert to making noise alert? It was perfect before, but now makes driving at night much more annoying (along with the spazzy "driver monitoring alert uncertain" alerts.

ShaneSmiskol, 2020-10-30:
Follow up; if this will not be changed, then I'd at least like to know if we can revert the driver monitoring times to their previous values without risking being banned. After all, they were good enough until the last commit.

geohot, 2020-10-31:
We can make the green alert appear sooner.

geohot, 2020-10-31:
Hmm, actually @ZwX1616, can we revert _AWARENESS_PRE_TIME_TILL_TERMINAL and _AWARENESS_PROMPT_TIME_TILL_TERMINAL? I'm not sure why we had to change that.

ShaneSmiskol, 2020-10-31:
Yeah that sounds about right, I felt like it gave me about 2 seconds before it made noise which wasn't ideal (you end up hearing that warning beep every ~30 seconds or so at night).

ShaneSmiskol, 2020-10-31:
I'll go out and test the original values/find ones that seem better as well as try to fix that uncertain alert flickering while the orange "touch wheel" alert is on screen and open a PR for that

ZwX1616, 2020-10-31:
> Hmm, actually @ZwX1616, can we revert _AWARENESS_PRE_TIME_TILL_TERMINAL and _AWARENESS_PROMPT_TIME_TILL_TERMINAL? I'm not sure why we had to change that.
> 
> Or @ShaneSmiskol, without changing _AWARENESS_TIME=35 if you find values for the others two that are better, we'd upstream.

Yeah we can do that. Something in the middle might be better? But yes I am okay with being more strict with the green while allowing more time to react before orange. The driver is expected to hold the wheel in that situation anyways.

ShaneSmiskol, 2020-10-31:
I went out to get a video of the second issue I described, you can see the uncertain warning keeps triggering since the orange alert keeps resetting it. I actually didn't expect it to disengage here, I thought since it wasn't showing either alert for a few seconds that it saw me and reset the timer but I guess it didn't.

https://streamable.com/xxx7on

ShaneSmiskol, 2020-11-03:
Have any ideas what can be changed to fix the flickering?

DaveyD233, 2020-11-03:
Adobe flash player should be discontinued in December, so as a result, we're going to have to make things more adaptable to HTML5 and webm devices or programming :-) just saying

naomiplays332, 2020-11-03:
what

> On Nov 3, 2020, at 8:05 AM, DaveyD233 <notifications@github.com> wrote:
> 
> —

ZwX1616, 2020-11-04:
> Have any ideas what can be changed to fix the flickering?

A counter can be added to avoid throwing multiple alerts within a short amount of time, but it's kind of hacky tbh. When it says "Check Driver Visibility", check driver visibility. If driver visibility can't be achieved, vision DM will not work as desired. I would just touch the wheel to reset.

ZwX1616, 2020-11-04:
That being said, you are welcome to find out a comfortable green/orange balance and submit a PR for that!

ShaneSmiskol, 2020-11-05:
I don't think not changing behavior is a good solution since it's especially confusing to know what the state of awareness is if it flickers on and off and then goes terminal out of nowhere. It was completely unexpected for me since it seemed like it picked up my face and was happy, but then gave me less than a second to react before disengaging.

I'll make a PR for updating the times though soon, I just need time to experiment

closed 2020-11-12 by ZwX1616