commaai/openpilot #2473: Double Brake when Approaching Stopped Vehicle

C0mpl3t3N00b, 2020-11-02:
**Describe the bug**

With OpenPilot enabled, when approaching a stopped car at a traffic light or heavy traffic situation, OpenPilot will slow to a stop behind the stopped car, but will then release the brakes and move forward slightly and stop again. This is with the lead car not moving at all.

**How to reproduce or log data**

Approaching a stopped vehicle will exhibit this behavior about 50% of the time.

**Expected behavior**

When approaching a stopped car, OpenPilot should slow to a stop at an appropriate distance from the lead car and only move forward if the lead car moves forward.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: bdad50b8a1f9fa8e|2020-11-01--16-47-58--5
 - Route: bdad50b8a1f9fa8e|2020-11-01--16-47-58--5
 - Timestamp: https://my.comma.ai/cabana/?route=bdad50b8a1f9fa8e%7C2020-11-01--16-47-58&exp=1635827347&sig=huumgNlfxiHDVminmiuY9uGtn%2FL3BO%2FrcSWvX5BBpg4%3D&max=18&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fbdad50b8a1f9fa8e%2F6cd92beecc597e5fb5f53b068f0c3f7a_2020-11-01--16-47-58
 - Version: master-ci
 - Car make/model: 2020 Rav4 Hybrid LE

**Additional context**

kozuch, 2020-11-08:
Can this be because humans sometimes do this to and the model learned it from data?

grekiki, 2020-11-08:
No, longitudinal policy is not learnt.

zorrobyte, 2020-11-08:
This is old news, has been an issue since at least 0.5. This is why they are working on E2E long control.

timgudex, 2020-11-09:
I get this as well but it looks like it will touch the vehicle rather then stopping again.

pd0wm, 2020-11-09:
Your case seems related to https://github.com/commaai/openpilot/issues/1270. 

Does this PR fix it? Also let me know if stops are too rough now, then we need to come up with a different solution. https://github.com/commaai/openpilot/pull/2508

yufeng66, 2020-11-09:
have exactly the same issue with my 2020 rav4 hybrid. There is one thing I noticed might be related. When the openpilot stops the car, the power flow display still shows the power flowing from the battery to the wheels.  If I stop the car by holding on to the brake, there is no power flow what so ever.

cydia2020, 2020-11-09:
> have exactly the same issue with my 2020 rav4 hybrid. There is one thing I noticed might be related. When the openpilot stops the car, the power flow display still shows the power flowing from the [...]

Usually that's due to the SnG hack

C0mpl3t3N00b, 2020-11-15:
Exhibited same issue again on new Torch-Model: 

bdad50b8a1f9fa8e|2020-11-14--12-08-53--2

RightTrash, 2020-11-27:
Have this issue at every stop with my 2020 rav4 hybrid xle.  
Sometimes it creeps a bit too, after that initial double brake (stop, sudden creep to stop again, then creep more).
More often than not it's fine and doesn't but the double braking seems possibly like a sketchy thing, if it were to brake hard then do that in some close call situation/s.

adeebshihadeh, 2020-12-17:
Is this still an issue after merging #2716?

mattbsea, 2021-01-05:
> Is this still an issue after merging #2716?

2020 RAV4H here, had the issue before version 0.8.1 but not since.  I believe it is fixed.

closed 2021-01-05 by pd0wm