commaai/openpilot #2474: After First update Comma Two V0.7.9 Stoped Working Two massages 1.Key chain has stopped

GBenke, 2020-11-02:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

VirtuallyChris, 2020-11-02:
Please reflash NEOS using this guide. https://github.com/commaai/eon-neos#restoring-on-linuxos-x If the issue still persists, email support@comma.ai

closed 2020-11-02 by VirtuallyChris

skyej944, 2021-09-12:
File "flash.sh", line 8
    if [ ! -f $FASTBOOT ]; then
         ^
SyntaxError: invalid syntax

skyej944, 2021-09-12:
This is what I get when running flash.sh