commaai/openpilot #2475: Encoder Dropped Frames

geohot, 2020-11-03:
**Describe the bug**

fcamera.hevc has 1126 in segment 0 and 1076 in segment 1.

**Expected behavior**

1200 frames. Also a single copy of encodeIdx in the log.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 02c45f73a2e5c6e9
 - Route: 02c45f73a2e5c6e9/2020-10-31--15-32-56/1
 - Version: master
 - Car make/model: Hyundai Sonata 2020

ZwX1616, 2020-11-03:
it happens when git is still fetching when going onroad according to @adeebshihadeh. seemingly only affects master/-ci. killing updated when car starts should fix this.

geohot, 2020-11-03:
I would rather not kill updated if we don’t have to. What’s loggerd priority now? If it’s nothing, I think that’s the real solution.

ZwX1616, 2020-11-03:
both loggerd and ui are pretty low right now.

adeebshihadeh, 2020-11-08:
loggerd now has minimum RT priority, more than android processes but less than the rest of the openpilot processes. 8f44e695835edbba79b913607d73f7735df060eb

closed 2020-11-08 by adeebshihadeh