commaai/openpilot #2493: Comma 2 Manuel Power on will not work after 30 hour no drive inactivity

patcoghm, 2020-11-06:
**Describe the bug**  
  Comma 2 can not be manually powered on after Comma2 was automatically powered off when the 30 hours of no new drive inactivity period has lapsed.  Before version 0.7.10 and maybe 0.7.9, Comma 2 could be powered on and rebooted after this 30 hours and after not driving every day.  The time out period was to save car battery drain, as a hybrid its critical as the 12v start battery is not a powerhouse device. Now it does nothing, no buzz, blue light steady blinking from panda, using stock version 0.7.10 with the new neos.  No forks just a "plain Jane" setup and auto installs.

  
  The car will not boot using the manual power button after auto power off inactivity..  The only way to get it booted is to start the car, this starts a boot from scratch, not from the normal screen shown when ready and running.

**How to reproduce or log data**  ..................
   We need to test of a Comma 2 in a car setup and power supply, with a drive completed, Then park it  for 30 hours or until it indicates "appears offline" from the Comma Prime app.  Tapping the screen would open OP ready screen if its less then the 30 hours.  Using stock 0.7.10, no forks, and then attempt to start Comma2 manually with the power button when Comma2 indicates its fully powerd off by the timer.
    It normally did power on and boot with a buzz. Now nothing and this would appears to be a software change or bug as its suppose to boot manually. I think the panda which is always running is not allowing it to boot due to a missing line of software.  Otherwise you need to start the car or it remains off line.   But because of the errors from the car it
[image]
 need to be turned off then restarted

  
  Time out the inactivity timer which has been said to be 30 hours of no drives or starting the car.  The Comma2 is installed and never removed and has been running for months and no cables are unplugged.  The device is not removed nightly as some users as I garage it.  So they might miss this bug as the reattaching Comma2 will poeron and boot automatically.

**Expected behavior**  After the 30 hours a push of the Comma 2 power button and Comma 2 did start a boot from power off.

  As per a discussion with Discord and customer support, they also think it is suppose to boot up.

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two].... 
   Comma 2

 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
   e4a16227cff292b5

 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
   after any driver, after a 30 hours of no drive which turn Comma2 off.

 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
    Any time after 30 hours and stuo power off of Comma2.
    
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
    possibly started in version 0.7.9 as I drove daily, but this version 0.7.10 with days of no driving, no forks, auto updated by your pushes.  Its worked since I got the device without any problems.  As for operation, Comma2 works as advertised and never had any cable problems or needed to insert them to solve any problems.

 - Car make/model: [e.g. Toyota Prius 2016]
   2020 Prius Prime XLE, un plugged DSU and in version 0.7.9 was when I started to try the unplugged DSU driving.  Unplugged DSU allows full Openpilot and the long (Speed and braking are far smoother).  I want to stay unplugged as its better then Toyota stock which is slow acceleration and hard and jerky braking which is truly their AEB at work.

**Additional context**
   Customer service advised it could be the OBD2 power unit or a plug.  I have a spare power module after its case opened and they sent me a second power module, so both were used.  I ran the 30hour time out power off using both power supplies this past week.  All cables are seated and are checked for firmness and fit.

    This manual booting allowed cars with unplugged  DSU's to avoid the car errors thrown when the car is started before Comma2.  Hence the errors,
"Pre-Collsion System Malfunction", 
"Headlight System Malfunction"
"Lane Departure Alert Malfunction"
These erros do not appear and all systems work if the Comma2 is ready before the car starts, and the reason for this bug report.

Attached a video, not professional showing the problem, and why allowing the Manual boot with the Comma2 power button is a necessary step before car starting for unplugged DSU users.

patcoghm, 2020-11-07:
Again the real point is the manual boot was the way for un plugged DSU users to avoid the car errors.  Starting the car first and booting Comma2 at the same time is a bad way to solve a problem.  The solution is start Comma2 first after a screen tap and its off, then the manual boot first and start the car when Comma2 is at its ready screen.  No errors and it allows Full Comma2 control.

robbederks, 2020-11-09:
We see what the issue is. To avoid power draw on EONs, we turn off the power to the USB port. 
This interface controls the power to the phone in the comma 2's, which indeed has the symptoms of not being able to turn on the device with the manual power button after the shutdown.
As far as I can tell by quickly looking at it, this logic has never been different (except that the 30h limit was changed to a smarter algorithm: #1994 ), but it's an annoying bug nonetheless.
We suspect the phone in a C2 doesn't suffer from the same power draw as EONs, so we'll change the logic there.

closed 2020-11-10 by robbederks