commaai/openpilot #2501: Carla sim "Openpilot unavailable Communication issue between processes"

Saumya-Shah, 2020-11-07:
I have ubuntu 18.04 and have run the two given scripts for running openpilot with Carla. I get a warning saying "Grey Panda is deprecated" followed by  "Openpilot unavailable Communication issue between processes" when I engage by pressing 1 on the terminal.
Operating system: [e.g. Ubuntu 18.04]

dorkmo, 2020-11-08:
theres currently a hack to remove it in controllerd

https://github.com/commaai/openpilot/wiki/CARLA#communication-issue-between-processes

join the discord server and check out channel #openpilot-simulation  a few people have had the same issue

Saumya-Shah, 2020-11-08:
Did you mean controlsd.py file? I removed the lines corresponding to radar commissue and commissue events in lines 210-214. That doesn't seem to fix it though, can you help? I have joined discord server but don't see more description regarding this over there. Thanks a lot. 

[image]

dorkmo, 2020-11-08:
sorry im not sure, davidtena98 most recently experienced it.  They start talking about it here... https://discord.com/channels/469524606043160576/684457101455786017/774270397243326474

LoganxDev, 2020-11-12:
#2528

Saumya-Shah, 2020-11-12:
Hi, thank you for the reply. Unfortunately, I still have the issue. The am running it as follows:

`sudo ./tools/sim/start_carla.sh`

Then in other terminal:
`sudo ./tools/sim/start_openpilot_docker.sh `

And then I change the following inside the docker container:

In bridge.py: I changed hwType from "greyPanda" to "blackPanda" as in  #2528 and 
I also comment the issue lines in controlsd.py as shown in the screenshot above. 

And then I rerun the bridge.py. 

Did I miss something?

LoganxDev, 2020-11-13:
It sounds like you didn't kill and restart openpilot. Looking at your screen shot if you go into the right hand terminal window then go to the first tmux window with ```` ` + 0 ````, then `ctrl + c` to kill openpilot. Make your edits then run `./launch_openpilot.sh`, finally you can swap back to the tmux window you started in with ```` ` + 1```` and rerun `bridge.py`. These steps are performing `tmux_script.sh` which is what `start_openpilot_docker.sh` is doing inside the docker container.

gucasbrg, 2020-11-15:
changed hwType from "greyPanda" to "blackPanda" as in #2528,
also wrong config
[image]

gucasbrg, 2020-11-30:
There are three interesting tips to improve

- The greypanda written in bridge.py caused thermald started false. Then, as soon as the manager was opened, all services were killed.
- When sim is used, all service alives are also monitored, and then there is even a frontframe, which is nothing. . .
- That is, when onnx is running on the cpu, the fps prediction is very low, and then the prompt is not very friendly

DanielYEHsieh, 2021-01-05:
Hi @adeebshihadeh,

thanks for replying.

I did the same change in
https://user-images.githubusercontent.com/22060625/98455550-2e19d300-2140-11eb-9deb-454dfce0791d.png
but it seems not fixed the issue,

and in bridge.py: is already changed from "greyPanda" to "blackPanda"
but it still runs into the same issue.

kill and restart openpilot did not help too.

Is the fix in discord?

Thanks!

PR #19824 (2021-01-19): Carla fixes
Added some fixes for known carla issues along with other patches:
- fixed issue with carla window not showing when CARLA is ran from docker
- added WASD keys for gas/brake/steering when running bridge.py
- comm issues on controldsd.py, this perhaps fixes #2501 
- CARLA camera provides a frame number https://carla.readthedocs.io/en/0.9.7/cameras_and_sensors/
closed 2021-01-19 by adeebshihadeh