commaai/openpilot #2504: Feature Request : Sliding Scale for Long, Lat Controls

jcwillia1, 2020-11-08:
I would love sliders to fine tune control :

drive a bit more to the right or a bit more to the left.  The Sonata stock centering drives just a touch too far to the right for my taste.

Acceleration curves from very aggressive to very conservative - I want my car to accelerate like a golf cart (aka, very VERY slowly)

Braking curves from very aggressive to very conservative (this is there somewhat with following distance in stock "smart cruise" today).  For my purposes in normal traffic, I always want the car in front of me to be moving away from me, not following at a set distance.  This is obviously not possible in stop and go traffic.

pd0wm, 2020-11-09:
We won't be adding sliders for this. As we slowly move to more end-to-end learned longitudinal policies it should start driving more like a human. For comma it's not worth it to spend time on these features if they are going to be replaced anyway.

Feel free to fork openpilot and change the values to your liking in the code. e.g.:
https://github.com/commaai/openpilot/blob/master/selfdrive/controls/lib/planner.py#L32

closed 2020-11-09 by pd0wm

jcwillia1, 2020-11-09:
why close the thread?  I believe there is room for a more nuanced
conversation here.