commaai/openpilot #2521: openpilot feedforward function doesn't match real world data

ShaneSmiskol, 2020-11-11:
So I was trying to gather data to make a script that could get the optimal feedforward k_f tuning value based on torque and angle and speed, which it did with least squares/scipy's `curve_fit` function. However when plotting the data I noticed that the `speed ** 2` part of the function didn't align with data plotted from my car. And sure enough, I felt this in my car as well; a value would steer well at 30 mph but oversteer at 70 mph, and vice versa if I tuned at 70 mph (understeer at 30 mph).

You can see from the bottom two graphs the orange line (current op ff function) is applying too much torque at 65 mph and not enough at 25 mph with a single value.

[image]

I then fitted a model that took in a k_f for angle, and 3 degrees of a polynomial for speed, and fitted against the data (speed and angle as input, torque as output). You can see the purple lines much more closely fit to the data plotted from my car.

(And yes, I offset the torque by my steer delay in the data so that what torque was being commanded matched with the angle that openpilot expected to reach).

Here's the function I fit:

```
def _custom_feedforward(_X, _k_f, _c1, _c2, _c3):  # automatically determines all params after input _X
  v_ego, angle_steers = _X.copy()
  steer_feedforward = angle_steers * (_c1 * v_ego ** 2 + _c2 * v_ego + _c3)
  return steer_feedforward * _k_f
```

Is there any chance of getting this upstreamed into openpilot? It really does seem smoother for me, and one other who's tested it says it feels better than stock too.

ShaneSmiskol, 2020-11-11:
My next step is going to be able to get these values from a cabana drive, rather than gathering data while in the car, so other cars can more easily be tuned. That would be the hardest part, updating and tuning every car currently in openpilot.

geohot, 2020-11-12:
We are certainly interested in this. Submit a PR for a model, and get a couple people to confirm it's better. I think a third order polynomial is a reasonable choice, though I don't know enough controls to say for sure.

ShaneSmiskol, 2020-11-12:
The only issue I'm having with getting the LogReader to work with the rlogs is that there seems to be lag in the logs? I use data from three different messages, pathPlan (angle_steers_des to filter out samples where angle isn't near it), carState, and can (for torque and if steer req). Since two are at 100Hz and one is 20Hz the data doesn't match up completely, and the time from carState to the can message grows from 0.001 to 0.2 seconds at times. Is there an easier way to compile data from multiple messages? Iterate through all messages at each point in time?

ShaneSmiskol, 2020-11-13:
Yeah the polynomial gives the well needed torque at low speeds, v_ego**2 just wasn't a good fit for all speeds. So is that equation just not what it's supposed to be; `realigning tire momentum (~ lateral accel)`? Or is lateral accel not what feedforward needs to be?

pd0wm, 2020-11-13:
v_ego**2 should mostly match the physical properties of the realigning force on the tires. But there are probably other things going in the EPS where it tries to compensate for speed leading to the observed behavior.

For the top plot I'm not entirely sure the new function is a better fit. They both look quite arbitrary given the data. I think a statistical test would fail horribly on both FF models. How repeatable is your fitting tool (how much do the values differ if you fit on two different routes)?

I also want to look into an on-line tuning of INDI for all cars. If that's very stable that should also solve a lot of problems.

ShaneSmiskol, 2020-11-13:
You're right about the top plot, but I still think the results from the bottom two (how the current function compares at low speed vs. high speed) are something to consider. Even while tuning I could tell that one value didn't work at all speeds, if I tuned kf at 30 mph for sharp curves, it would always oversteer slightly at 60+ mph.

So with the data from the rlogs using the MultiLogIterator, the same exact drives give much noisier data compared to my initial data gathering (writing to a file from carcontroller with everything I needed), for example there's a considerable number of samples where torque is > 1000 and yet angle is < 5 deg. These samples don't show up in the data I gathered myself, using the same exact filtration code. So I decided to use the speed poly from my drives and just have the script fit k_f, replacing `v_ego ** 2` with the poly function I fit from my custom data (50+ miles).

I think it at least needs a few miles engaged with curves, just one route won't be enough data to give an accurate result, even just fitting k_f, not the poly for each car.

ShaneSmiskol, 2020-11-13:
By the way, here's the poly I found from ~50 miles of data, nearly always engaged with sharp curves:

```
def _custom_feedforward(_X, _k_f):
  v_ego, angle_steers = _X.copy()
  _c1, _c2, _c3 = 0.34365576041121065, 12.845373070976711, 51.63304088261174
  steer_feedforward = angle_steers * (_c1 * v_ego ** 2 + _c2 * v_ego + _c3)
  return steer_feedforward * _k_f
```

zorrobyte, 2020-11-14:
`I also want to look into an on-line tuning of INDI for all cars. If that's very stable that should also solve a lot of problems.`

INDI is pretty amazing, 0.1 timeconstant and 0.5 cost on TSS1.5/TSS2 (torque angle sensor) is a dream.

ShaneSmiskol, 2020-11-20:
I think there is a fundamental issue with INDI that might only be a problem on older cars (especially TSS1 with its worse angle and torque sensor/motor). I notice that if I'm on a bumpy road with pot holes that while the wheel shakes INDI does not apply any torque whatsoever, causing me to leave the path and eventually the road if it goes on for long enough. I can replicate this behavior on a smooth road at any speed if I just shake the wheel back and forth slightly, I can pull the car out of the lane without any resistive torque against me. When I let go, it takes half a second or so for it to recover, which can be very dangerous.

With PID, while shaking it/going over bumpy road it always tries to apply torque and keep me on path. When letting go after shaking it immediately recovers.

I tried tuning INDI for almost an hour and couldn't get anything nearly as good as PID+F. Maybe it's my car, or my bad tune, but INDI just isn't very safe for my Corolla on bumpy road (or anything external that moves the wheel like camber as well). Just wanted to share an anecdotal experience I had with INDI.

closed 2020-11-20 by adeebshihadeh