commaai/openpilot #2536: Veering into turn lane and exit while car is turning

alancyau, 2020-11-12:
**Describe the bug**

Car has the tendency to veer into the turn lane or exit while the car is the middle of a turn.

**How to reproduce or log data**

This behavior tends to happen during the middle of a turn. The car seems to want to hug the inside of the turn, causing it to veer into turn lanes or exits.

**Expected behavior**

Car should remain in the center of the lane and not veer off into the turn lane or exit.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 5208d2146d7f0093

 - Route: 
Veering into turn lane: 5208d2146d7f0093|2020-11-12--16-33-02--5
Veering into exit: 5208d2146d7f0093|2020-11-12--16-01-36--7

 - Timestamp: 
Veering into turn lane: 16:38:38
Veering into exit: 16:09:23

 - Version: master-ci 0.8
 - Car make/model: Honda CRV 2017 with EPS mod

**Additional context**

N/A

HaraldSchafer, 2020-11-19:
Thanks for the report. We are trying to move away from using lanelines for pathplanning. As we progress with this, behavior should improve.

HaraldSchafer, 2020-11-19:
similar to https://github.com/commaai/openpilot/issues/2575

alancyau, 2020-11-30:
I noticed this issue hasn't occurred again in the recent "gold" 0.8 release.

closed 2021-03-03 by HaraldSchafer