commaai/openpilot #2539: Unable to remain in center of lane in low speed

TriHoangSG, 2020-11-13:
**Describe the bug**

C2 unable to to remain in center of lane in low speed, light rain condition

**How to reproduce or log data**

Segments:

f72a3ad7dc38d5cf|2020-11-12--16-17-44--20

f72a3ad7dc38d5cf|2020-11-12--16-17-44--25

f72a3ad7dc38d5cf|2020-11-12--16-17-44--27

f72a3ad7dc38d5cf|2020-11-12--16-17-44--35

**Device/Version information (please complete the following information):**
 - Device:  Comma Two
 - Dongle ID: f72a3ad7dc38d5cf
 - Version: 0.7.10
 - Car make/model: Honda CR-V 2018

**Additional context**

Openpilot enabled to follow slow highway traffic in light rain condition

VirtuallyChris, 2020-11-14:
To clarify; is this because your car cuts steering at 12mph? Or is it some model bug?

pd0wm, 2020-11-16:
Even at speeds of 30mph the steering still doesn't seem to respond to torque commands. Is this always the case for you?

TriHoangSG, 2020-11-16:
> To clarify; is this because your car cuts steering at 12mph? Or is it some model bug?

I don't think the car cuts the steering, i think it's a model bug, it doesn't see that it's not centered and not correcting itself.

TriHoangSG, 2020-11-16:
> Even at speeds of 30mph the steering still doesn't seem to respond to torque commands. Is this always the case for you?

At speed, the steering doesn't respond well in tight curves on the highway. I was driving on the I95 and CT15 highways between NYC and Bridgeport (CT) and whenever there is a tight curve (with warning signs indicating tight curve on the road) it always warn 'turning exceeds limit' and can't make the turn.

TriHoangSG, 2020-11-16:
See this segment: f72a3ad7dc38d5cf|2020-11-13--18-46-48--100

Here is coming out of JFK airport, lane detection is so bad, the lane is clearly visible.

pd0wm, 2020-11-16:
@TriHoangSG In the segment from the bug report it's not a model problem. It's sending torque, but the wheel is not turning.

rav4kumar, 2020-11-18:
this happened to me as well. on 0.8 but it doesnt happened on the 0710

dorkmo, 2020-11-18:
be sure to share a segment!

adeebshihadeh, 2021-03-09:
Is this still an issue?

VenomrushCT, 2021-03-23:
> Is this still an issue?

This is still an issue in v0.8.2
More noticeably when the car comes to a complete stop, then move again with traffic

adeebshihadeh, 2021-06-18:
Do you have some more recent segments? I suspect this is a panda issue.

pd0wm, 2021-10-06:
@VenomrushCT Is this still an issue? If so any recent routes with the problem?

VenomrushCT, 2021-11-24:
I was in heavy traffic this past weekend and did not see it ping pong on 0.8.10 with lane line so we can close this issue

closed 2021-11-24 by VenomrushCT