commaai/openpilot #2545: 0.8 - Yellow lead indicator isn't in 3D

ShaneSmiskol, 2020-11-13:
Whenever I'm going up or down a hill with a lead, the yellow indicator doesn't follow the car and stays in its current y position.

VirtuallyChris, 2020-11-14:
I noticed this and was about to report it. Nice job 😆

HaraldSchafer, 2020-11-16:
Ah yes, I can fix that

HaraldSchafer, 2020-11-17:
@deanlee Interested in looking into this? The lead prediction of the model only predicts and x and y, no z (height). But we could use the height of predicted path at that point to draw the lead triangle.

deanlee, 2020-11-17:
@HaraldSchafer  good idea. I'd like to have a try in pull request https://github.com/commaai/openpilot/pull/2525

adeebshihadeh, 2021-01-16:
@deanlee can we bring this out of the big UI PR and make a small PR for just this?

deanlee, 2021-01-17:
@adeebshihadeh added 20 lines of code in this new PR: https://github.com/commaai/openpilot/pull/19793

closed 2021-01-28 by pd0wm