commaai/openpilot #2546: Master-ci 0.8.0 Model Lagging Error

nickgeurnop, 2020-11-14:
Issue:

Model Lagging error on 0.8.0 pre release. Normal bootup, no dash faults

Details:

 - Device: comma two
 - Dongle ID: 1615388ccbe3edc9
 - Route: 1615388ccbe3edc9|2020-11-13--19-11-34
 - Timestamp: About 2 minutes into drive, repeated lagging until reboot
 - Version: master-ci 0.8.0 pre-release
 - Car make/model: 2019 Toyota Corolla Hatchback XSE (TSS2.0)

Updated, restarted C2 and car. Issue went away. Will monitor and report.

pd0wm, 2020-11-16:
Looks like loggerd was using more CPU than normal. @adeebshihadeh any thoughts?

nickgeurnop, 2020-11-16:
Issue:

Model Lagging error on 0.8.0 pre release. Normal bootup, no dash faults

Details: 

Happened twice in quick succession. Was able to re-engage a minute later unlike first bug report. 

Expected Behavior:

Model does not lag and does not throw error forcing disengagement.

Device: comma two
Dongle ID: 1615388ccbe3edc9
Route: 1615388ccbe3edc9|2020-11-16--09-25-09
Timestamp: @ 9:31:16 
Version: master-ci 0.8.0 pre-release
Car make/model: 2019 Toyota Corolla Hatchback XSE (TSS2.0)

nickgeurnop, 2020-11-18:
Issue:

Model Lagging error on 0.8.0 pre release. Normal bootup, no dash faults

Details:

Happened twice in quick succession. Was NOT able to re-engage a minute later like first bug report.

Expected Behavior:

Model does not lag and does not throw error forcing disengagement.

Device: comma two
Dongle ID: 1615388ccbe3edc9
Route: 1615388ccbe3edc9|2020-11-18--17-35-44
Timestamp: @ 17:47:30, many times afterward for about 10 minutes
Version: master-ci 0.8.0 pre-release
Car make/model: 2019 Toyota Corolla Hatchback XSE (TSS2.0)

nickgeurnop, 2020-11-18:
I'd like to mention that I am consistently updating the Master-ci branch while reporting these bugs.

adeebshihadeh, 2020-11-19:
Your last route isn't uploaded. Can you connect your device to wifi until all the logs are uploaded?

nickgeurnop, 2020-11-19:
I made the report maybe 15 minutes after getting home. Please let me know if it is still unavailable. C2 connects to wifi from driveway when I get home + I have my 3 months of comma prime.

geohot, 2020-11-20:
Can confirm DM model is taking ~37 ms to run when it usually takes ~14 ms on 1615388ccbe3edc9/2020-11-13--19-11-34/0

1615388ccbe3edc9/2020-11-15--12-38-37/1 is running in the usual 14.

geohot, 2020-11-20:
Fix for dmonitoringmodeld CPU usage:

https://github.com/commaai/openpilot/commit/826c21b045b09d6e28b313b3c57cd62800febb16

Though aside from the wasted RAM bandwidth, I'm not sure how this causes modeld lag. Maybe it is just the RAM bandwidth?

adeebshihadeh, 2020-11-20:
@nickgeurnop This should be fixed in master-ci now. Re-open the issue if it happens again after updating.

closed 2020-11-20 by adeebshihadeh