commaai/openpilot #2547: Add replay (unlogger) to QT UI

geohot, 2020-11-14:
So at first it sounds crazy. But then you remember how bad unlogger is and you think about it more.

A dashcam has the ability to replay the "recording" on the device. openpilot should too, and it can replace unlogger. We do a really bad job surfacing the ability to replay routes in the detail we record them with the tooling and app.

This completes the device as a dashcam, but...

The real win would be for PCs. I run ui on my PC, pair the PC to my account, and can access all the routes from my device. I replay one, I can inspect the messages, replace services, etc...all the things we do now with unlogger.

The docker is great for carla (minus the current bug). I was pleasantly surprised when the openpilot UI came up. Imagine I could, with just the one docker command and a quick pair to my account from my phone, start replaying a drive.

pd0wm, 2020-11-16:
Would be nice if we can also replace ui.py. The only reason I still use that is for plots of speed and steering angles. If we can add a second window to the qt ui to plot/visualize arbitrary cereal messages that would be a great improvement over ui.py

marschr, 2021-01-20:
Did a mockup of an c2 version of it..
[image]

Was the `tools/nui` thingy ever related to this idea? It seemed like some QT version of unlogger.

adeebshihadeh, 2021-04-20:
TODOs
- [ ] initial headless PR (#20602)
  - [ ] streaming + socketless mode
  - [ ] fetch routes with the public API
  - [ ] all python unlogger functionality
- [ ] nice route selector in the UI
- [ ] smart segment fetching
- [ ] works well on-device
- [ ] nice demo route with docker?

adeebshihadeh, 2021-12-15:
replay is good now, but do we still want this? I don't see this becoming better than connect/useradmin, and it's tons of code to add to the UI.

deanlee, 2021-12-23:
Maybe a basic version can be implemented without too much code, I will try to see if it works

deanlee, 2021-12-23:
it may not require a lot of code to only implement basic functions. let me have a try.

closed 2024-06-20 by adeebshihadeh