commaai/openpilot #2551: Parked car on the side of the road

Mohd5566, 2020-11-14:
Hi 
I noticed the comma ai2 does not recognize parked car on the side of the road. If I didn't hold the steering it would drove through them.

Comma ai two cd226019363ea148

TriHoangSG, 2020-11-18:
> Hi
> I noticed the comma ai2 does not recognize parked car on the side of the road. If I didn't hold the steering it would drive through them.
> 
> Comma ai two cd226019363ea148
> Toyota Highlander 2020

can you share the drive segment?

Martint1980, 2020-11-18:
Have the same problem, but my drives are already deleted

Mohd5566, 2020-11-18:
> > Hi
> > I noticed the comma ai2 does not recognize parked car on the side of the road. If I didn't hold the steering it would drive through them.
> > Comma ai two cd226019363ea148
> > Toyota Highlander 2020
> 
> can you share the drive segment?

I have the driving video on Comma ai. How can I share it?

Mohd5566, 2020-11-19:
11/17/2020 5:30 To 6:03 last 04:00 minutes where cars parked.

TriHoangSG, 2020-11-19:
> I have the driving video on Comma ai. How can I share it?

Go to https://my.comma.ai
Open the drive with issue
Seek to where the issue occurs
Click the 'Copy Current Segment' found at the bottom right of the page
Paste the details here, format will be something like f72a3ad7dc38d5cf|2020-11-18--15-57-59--0

HaraldSchafer, 2020-11-19:
Thanks for the reports. We are trying to move away from using lanelines for pathplanning. As we progress with this, behavior should improve.

Mohd5566, 2020-11-19:
> Go to https://my.comma.ai
> Open the drive with issue
> Seek to where the issue occurs
> Click the 'Copy Current Segment' found at the bottom right of the page
> Paste the details here, format will be [...]

Drive segment 
cd226019363ea148|2020-11-19--17-19-59--32

HaraldSchafer, 2021-03-31:
Hopefully improved with laneless.

closed 2021-03-31 by HaraldSchafer