commaai/openpilot #2554: No Data From Device Sensors

gbergman14, 2020-11-14:
Appears soon after car starts - before Openpilot becomes fully ready.
message says to "reboot device", no help
Hard reset doesn't fix it either. 

First occurred with Kegman fork 0.7.8. Upgrade to 0.7.10 didn't help
Message was "No Data from EON Sensors"

After hard reset, installed stock OpenPilot 0.7.10
Message now reads: 'No Data From Device Sensors" Reboot your device.

Unplug Gray Panda, reset EON, restore everything (Stock OpenPilot): no luck

**Device/Version information (please complete the following information):**
 - Device: EON Gold
 - Dongle ID: a956eb1a59e8acda

 - Route: n/a
 - Timestamp: n/a
 - Version: 0.7.10 release2
 - Car make/model: Volt 2017 - Grey Panda. Vasilly Giraffe

Everything was working OK until recently.
Tried searching discord and github. Couldn't find any references.

Any ideas?
Gary Bergman
garyisfinally -at- gmail
732-247-2727

pd0wm, 2020-11-16:
Confirmed there is no sensor data in your latest drive. Sounds like a hardware issue if persists after hard reset and switching back to release2.

gbergman14, 2020-11-16:
Any way to determine whether the problem is in the EON Gold or the Grey Panda?
Is there an upgrade path (discount) for either device?

VirtuallyChris, 2020-11-16:
Please refer to https://comma.ai/tradein for $300 off a comma two!

closed 2020-11-16 by VirtuallyChris

neshant, 2020-11-19:
@VirtuallyChris In fact I witnessed the same issue in my toyota RAV4 after I started the car after 3 days of not using it. I purchased the car 7 days back and it worked fine in the first 3 days , but suddenly I saw the same error. In order to fix this I turned off the car and pulled out the OBD connection with RJ45 underneath the steering wheel and plugged it back in which fixed it. I hope it does not happen again. I am just wondering what should I do if it happens again. My intuition is that since comma device shuts off itself after 36 hours , maybe that has something to do with that error.

pd0wm, 2020-11-20:
The no sensor data is something internal to the device. If it happens more often contact support@comma.ai and they will arrange a replacement of the comma 2.