commaai/openpilot #2555: 2018 Lexus IS 350 F Sport - mid drive CAN Error

kndonlee, 2020-11-15:
**OP Disengages mid-drive with CAN Error and loud beeping**

During an openpilot controlled drive, one of several failure scenarios occurs:

* OP shows CAN Error: check connections + loud triple beeping
* OP triple beeps then goes to status screen showing miles driven, comma points, etc.
* OP triple beeps, then reboots to being functional again
* OP triple beeps, then reboots to not recognizing vehicle.

**log data**

Reproduction steps is to drive a while, then failures occur.

2020-11-13 around 6:15p EST - triple beeped, then went to status screen. OP did not recognize car after restarting vehicle.

2020-11-14 around 1:24p EST - triple beeped, then went to status screen.

2020-11-14 around 3:36p EST - triple beeped, then displayed CAN Error: check connections.

**Expected behavior**

Doesn’t disengage and continues to operate.

**Device/Version information (please complete the following information):**
 - Device: comma2
 - Dongle ID: 268f0090ec7ae6a7
 - Route: 268f0090ec7ae6a7|2020-11-14--15-21-15
 - Timestamp: 2020-11-14 3:36pm EST
 - Version: 0.8.0 master-ci branch
 - Car make/model: 2018 Lexus IS 350 F Sport

**Additional context**

* if it’s a hardware issue, bumpy roads seem to affect error rates a bit.
* seems to run well on smooth highways for 10-30 min stretches before failing
* replacement harness box seems less reliable than original harness box
* could be related to another user with a 2017 is350 where v1 fingerprinting works for him but v2 fingerprinting does not. Is it possible to lose fingerprinting mid-drive?

pd0wm, 2020-11-16:
Can you connect your comma two to wifi so the full logs upload?

adeebshihadeh, 2020-11-20:
It seems very likely this is hardware. I'd recommend disconnecting all your hardware and reconnecting everything firmly. If the issue persists, reopen the issue with a fully uploaded route.

closed 2020-11-20 by adeebshihadeh