commaai/openpilot #2556: v0.8 Still Exhibits Poor Lane Detection of Black Colored Lane Markers

C0mpl3t3N00b, 2020-11-15:
**Describe the bug**

Continuation of https://github.com/commaai/openpilot/issues/2210

Lines on the highway are painted black (seem to be painted over) but lane reflectors are still present, and seem to have enough contrast that I and other drivers can tell where lane markers are . When Comma is activated it does not see lane markers and will sometimes ride over the lines or get very close to the lane marker it does not see.

**How to reproduce or log data**

Driving South Bound on Hwy 101 from Morgan Hill, CA to Gilroy, CA will reproduce this consistently.

**Expected behavior**

If I and all other drivers can see the lane dividers then Comma should also "see" the lane markers too, so that my vehicle can stay centered in the lane and not begin to ride over the line.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: bdad50b8a1f9fa8e
 - Route: bdad50b8a1f9fa8e_2020-11-14--12-08-53--28
 - Timestamp: Starts around 23 minutes into drive
 - Version: master-ci f6568592391fd71303d1b6ac2ec907c45a57f1cd
 - Car make/model: 2020 Toyota Rav4 Hybrid LE

**Additional context**

[image]

HaraldSchafer, 2020-11-19:
Thanks for the report. We are trying to move away from using lanelines for pathplanning. As we progress with this, behavior should improve.

geohot, 2020-12-11:
This was a bad definition choice, we are working on it now. See #comma-pencil on discord to come help. Hopefully fixed in 0.8.2

HaraldSchafer, 2021-03-31:
New segnet was used to train model for 0.8.3, should have improved behavior in these cases

closed 2021-03-31 by HaraldSchafer