commaai/openpilot #2563: Lexus IS fails fingerprinting

kndonlee, 2020-11-17:
Lexus IS fails to be detected by open pilot

 `car bug`

** Reproduce bug **

Plug comma2 on master-ci branch. Car is not detected and runs in dashcam-only mode.

**Expected behavior**

We expect the car to be detected and OP to be available for engagement.

**Device/Version information (please complete the following information):**
- Device: comma2
- Dongle ID: 268f0090ec7ae6a7
- Route: 268f0090ec7ae6a7|2020-11-16--08-32-58
- Timestamp: 2020-11-16 8:32am EST
- Version: 0.8.0 master-ci branch
- Car make/model: 2018 Lexus IS 350 F Sport

**Additional context**

It appears the addition of the 2017 Lexus IS git commit (`dc819a3df138f3dc39f81489d5bd539901eaf960`)
has led to a regression.

I temporarily fixed the detection by removing the following: 

```
root@localhost:/data/openpilot$ git diff
diff --git a/selfdrive/car/toyota/values.py b/selfdrive/car/toyota/values.py
index a352fa08..1c054b41 100644
--- a/selfdrive/car/toyota/values.py
+++ b/selfdrive/car/toyota/values.py
@@ -748,13 +748,6 @@ FW_VERSIONS = {
       b'\x018966353R1100\x00\x00\x00\x00',
       b'\x018966353R8100\x00\x00\x00\x00',
     ],
-    (Ecu.engine, 0x7e0, None): [
-      b'\x02353P7000\x00\x00\x00\x00\x00\x00\x00\x00530J5000\x00\x00\x00\x00\x00\x00\x00\x00',
-    ],
-    (Ecu.esp, 0x7b0, None): [
-      b'F152653330\x00\x00\x00\x00\x00\x00',
-      b'F152653301\x00\x00\x00\x00\x00\x00',
-    ],
     (Ecu.dsu, 0x791, None): [
       b'881515306400\x00\x00\x00\x00',
       b'881515306500\x00\x00\x00\x00',
```

Also the Ecu.esp section in the does not show up consistently in the carFw output, so the car or OP needs to be restarted several times in order for OP to recognize the vehicle.

See:  2020-11-15--20-10-20--0 (does not have ecu = esp) vs 2020-11-16--08-32-58--0 ( has ecu=esp)

```
    carFw = [
      ( ecu = esp,
        fwVersion = "F152653330\x00\x00\x00\x00\x00\x00",
        address = 1968,
        subAddress = 0 ),
      ( ecu = dsu,
        fwVersion = "881515306400\x00\x00\x00\x00",
        address = 1937,
        subAddress = 0 ),
      ( ecu = eps,
        fwVersion = "8965B53280\x00\x00\x00\x00\x00\x00",
        address = 1953,
        subAddress = 0 ),
      ( ecu = engine,
        fwVersion = "\x018966353R1100\x00\x00\x00\x00",
        address = 1792,
        subAddress = 0 ),
      ( ecu = fwdRadar,
        fwVersion = "8821F4702300\x00\x00\x00\x00",
        address = 1872,
        subAddress = 15 ),
      ( ecu = fwdCamera,
        fwVersion = "8646F5301300\x00\x00\x00\x00",
        address = 1872,
        subAddress = 109 ) ],
```

kndonlee, 2020-11-17:
Does the fingerprint v2 detection require that all sections are detected in order for there to be a match?  The Lexus IS comes in 2 engine trims: 4 cyl & 6 cyl.  @Melowec also has a 2017 Lexus IS 350 (6 cyl) but appears to have a sub-engine type that my 2018 IS 350 does not.

Do multiple Lexus_IS identities need to be created?

Melowec, 2020-11-17:
Apologies if my commit broke anything. I was trying to add both FPv1 and FPv2 data for my car for completeness sake, but upon commit review, they made me remove FPv1 data.

Now when I update to the version with just FPv2 data, my car is showing unsupported again.

Similar to @kndonlee my car also intermittently shows ecu = esp data. I tried removing all other data for CAR.LEXUS_IS and running just the parameters that matched my car +/- the esp. No luck. The only way I can get my car recognized is by using FPv1.

- Device: comma2
- Dongle ID: c69fed081ccdb05a
- Route: c69fed081ccdb05a|2020-11-17--13-02-38
- Timestamp: 2020-11-17 19:09:24
- Version: 0.8.0 master-ci branch
- Car make/model: 2017 Lexus IS 350 F Sport

Output 1
<pre><code>
    carFw = [
      ( ecu = dsu,
        fwVersion = "881515306200\x00\x00\x00\x00",
        address = 1937,
        subAddress = 0 ),
      ( ecu = eps,
        fwVersion = "8965B53280\x00\x00\x00\x00\x00\x00",
        address = 1953,
        subAddress = 0 ),
      ( ecu = engine,
        fwVersion = "\x02353P7000\x00\x00\x00\x00\x00\x00\x00\x00530J5000\x00\x00\x00\x00\x00\x00\x00\x00",
        address = 2016,
        subAddress = 0 ),
      ( ecu = fwdRadar,
        fwVersion = "8821F4702100\x00\x00\x00\x00",
        address = 1872,
        subAddress = 15 ),
      ( ecu = fwdCamera,
        fwVersion = "8646F5301200\x00\x00\x00\x00",
        address = 1872,
        subAddress = 109 ) ],
</code></pre> 

Output 2
<pre><code>
    carFw = [
      ( ecu = esp,
        fwVersion = "F152653301\x00\x00\x00\x00\x00\x00",
        address = 1968,
        subAddress = 0 ),
      ( ecu = dsu,
        fwVersion = "881515306200\x00\x00\x00\x00",
        address = 1937,
        subAddress = 0 ),
      ( ecu = eps,
        fwVersion = "8965B53280\x00\x00\x00\x00\x00\x00",
        address = 1953,
        subAddress = 0 ),
      ( ecu = engine,
        fwVersion = "\x02353P7000\x00\x00\x00\x00\x00\x00\x00\x00530J5000\x00\x00\x00\x00\x00\x00\x00\x00",
        address = 2016,
        subAddress = 0 ),
      ( ecu = fwdRadar,
        fwVersion = "8821F4702100\x00\x00\x00\x00",
        address = 1872,
        subAddress = 15 ),
      ( ecu = fwdCamera,
        fwVersion = "8646F5301200\x00\x00\x00\x00",
        address = 1872,
        subAddress = 109 ) ],
</code></pre> 

FPv1
<pre><code>
number of messages 130:
fingerprint 36: 8, 37: 8, 114: 5, 119: 6, 120: 4, 170: 8, 180: 8, 186: 4, 238: 4, 400: 6, 426: 6, 452: 8, 464: 8, 466: 8, 544: 4, 550: 8, 552: 4, 608: 8, 610: 5, 643: 7, 705: 8, 740: 5, 744: 8, 800: 8, 815: 5, 836: 8, 845: 5, 849: 4, 869: 7, 870: 7, 871: 2, 896: 8, 897: 8, 900: 6, 902: 6, 905: 8, 911: 8, 913: 8, 916: 3, 917: 5, 921: 8, 933: 8, 944: 8, 945: 8, 951: 8, 955: 8, 956: 8, 979: 2, 992: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1005: 2, 1008: 2, 1009: 8, 1014: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1043: 8, 1044: 8, 1056: 8, 1059: 1, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1168: 1, 1176: 8, 1177: 8, 1178: 8, 1179: 8, 1181: 8, 1182: 8, 1183: 8, 1184: 8, 1185: 8, 1186: 8, 1187: 8, 1189: 8, 1190: 8, 1191: 8, 1192: 8, 1193: 8, 1196: 8, 1206: 8, 1208: 8, 1212: 8, 1226: 8, 1227: 8, 1235: 8, 1237: 8, 1279: 8, 1552: 8, 1553: 8, 1554: 8, 1555: 8, 1556: 8, 1557: 8, 1561: 8, 1568: 8, 1569: 8, 1570: 8, 1571: 8, 1572: 8, 1575: 8, 1584: 8, 1585: 8, 1589: 8, 1592: 8, 1593: 8, 1595: 8, 1598: 8, 1599: 8, 1666: 8, 1667: 8, 1669: 8, 1728: 8, 1745: 8, 1779: 8, 1904: 8, 1912: 8, 1990: 8, 1998: 8
</code></pre>

VirtuallyChris, 2020-11-18:
@adeebshihadeh We'll have to  look into this before 0.8. I think what might be happening is that it's looking for both engine fw versions now but only finds 1 depending which car so it never fingerprints properly. https://github.com/commaai/openpilot/pull/2553/files

adeebshihadeh, 2020-11-20:
Fixed in https://github.com/commaai/openpilot/commit/27fefb9f608fa43bf6278be2b2bef1c34f1289be and tested against one of your routes.

closed 2020-11-20 by adeebshihadeh