commaai/openpilot #2571: Driving Model Request: Segmentation for Semi-Trucks or other Long Body/Cargo Vehicles

C0mpl3t3N00b, 2020-11-18:
The current longitude control of Comma has a follow distance that is mostly acceptable in given situations but the follow distance for large semi trucks puts the vehicle in the truck driver's blind spot. When following a semi truck I, as the driver using OpenPilot perfectly center in the lane, cannot see the side mirrors of the truck therefore the trucks driver is more than likely unable to see me. Riding in the blind spot of any vehicle is dangerous, and this danger is amplified when large trucks are involved at speed.

Suggestion: The model should be able to recognize large-vehicles (like trucks), and set a follow distance such that the side mirrors of the lead large-vehicle are visible to Comma.

pd0wm, 2020-11-19:
We won't be building any hand-tuned features for following distance. When we switch to end-to-end longitudinal it will keep following distance like a human would.

closed 2020-11-20 by HaraldSchafer

dorkmo, 2020-11-20:
maybe could put vehicles GVWR in the fingerprint.
Possibly could calibrate current weight based on brake force and speed deceleration, rpm and acceleration?

C0mpl3t3N00b, 2020-11-21:
> We won't be building any hand-tuned features for following distance. When we switch to end-to-end longitudinal it will keep following distance like a human would.

@pd0wm Most humans do not keep a following distance that is safe or appropriate, not just in a semi-truck scenario. With this logic, will the model also be modeled after human behavior of being poorly centered, veering into adjacent lanes, or wavering in the lane (e2e latitudinal) too?

HaraldSchafer, 2020-11-21:
Our idea of how to get superhuman driving from human driving is explained a little more here https://medium.com/@comma_ai/towards-a-superhuman-driving-agent-1f7391e2e8ec

It is also possible to make a model be more conservative than the people it is learning from without handcoding any policies.