commaai/openpilot #2572: Simulator CI test

adeebshihadeh, 2020-11-18:
The test should run openpilot in the simulator for a minute and fail on process crashes or unexpected alerts. Ideally, this test should be able to run in GitHub Actions.

ekdnam, 2020-11-21:
I would love to work on the issue. Can you assign me?

geohot, 2020-11-24:
No assignment required, just submit a PR!

ekdnam, 2020-11-25:
Okay sure! Can someone give a few guidelines on how to solve the issue? I am new to testing, and would love to get some insights on how to proceed

grekiki, 2020-11-27:
First get the simulator to work on your PC

zachwildd, 2020-11-30:
Looks like you'll have to use self-hosted runners to get a GPU

adeebshihadeh, 2020-11-30:
If we need to use a GPU, we can just run this on our existing jenkins machine, but it doesn't really need to run well. For now, we just want to check for crashes or regressions, like the grey panda deprecation breaking the sim.

gijskoning, 2022-03-20:
@adeebshihadeh I build a draft pr for carla and bridge.py connection in CI. https://github.com/commaai/openpilot/pull/24009 However, GPU is really needed according to the Carla docs. Do I need to create a similar setup in the Jenkins file?

adeebshihadeh, 2022-03-20:
@gijskoning if you write a test that runs on my PC, I"ll do the CI integration

closed 2022-05-06 by adeebshihadeh