commaai/openpilot #2575: [0.8] Car doesnt stay in lane when an additional turn lane is added

abdulrahman004, 2020-11-19:
Have read multiple people noticing this issue; but didnt see a bug report

Single lane road turns into two lane with additional lane for left turn. The lane edges widens and car tries to center in middle until it can find the lane marker that divides the lanes.

**Expected behavior**
The car should continue using the right lane edge as reference instead of the 2 lane widened edges

 - Device: FreEon
 - Dongle ID: 59f457ce47259dd2
 - Route: 59f457ce47259dd2|2020-11-19--02-21-02--2
 - Timestamp: 18:23:27
 - Version: d96d0485696ca46fdb52e465bea21cb09e40546c
a359d8a30f5323bba599df32c70ae02863e5bf21 
 - Car make/model: Toyota Sienna 2018

**Additional context**

gif of the issue
https://imgur.com/a/c4JmkQh

HaraldSchafer, 2020-11-19:
Thanks for the report. We are trying to move away from using lanelines for pathplanning. As we progress with this, behavior should improve.

HaraldSchafer, 2020-11-19:
closed and linked to https://github.com/commaai/openpilot/issues/2536

closed 2020-11-19 by HaraldSchafer