commaai/openpilot #2579: [0.8] Face detection issue - constant warning

wongjunjin, 2020-11-20:
Face detection has been much more harder to detect my eyes / face. Even when I stare in front / right at the camera, it shows that "Driver is distracted". I am on right hand drive. 

**Expected behavior**
0.7.8 was better at face detection without the constant warning. 

**Device/Version information (please complete the following information):**
 - Device: comma two 
 - Dongle ID: 7bc774bc8e17fd8a
 - Route: 7bc774bc8e17fd8a|2020-11-18--15-27-50
 - Timestamp: Throughout drive
 - Version: 0.8.0
 - Car make/model: Toyota Corolla 2019

**Additional context**

ShadowVlican, 2020-11-20:
DM at night has been crap since 0.7.8.... i had no issues using my comma2 at night with 0.7.7

wongjunjin, 2020-11-20:
I just tried with 0.7.10. And it works great though. Really something to look at for 0.8.

ZwX1616, 2020-11-20:
> Timestamp: Throughout drive

Are you sure they were DM alerts? Just check the whole route and only found a few seconds of green alerts in segment 9, other than that nothing DM related. I do see a lot of LDW warnings throughout the drive though.

adeebshihadeh, 2020-11-20:
Feel free to reopen with a segment with frequent DM alerts.

closed 2020-11-20 by adeebshihadeh