commaai/openpilot #2584: Hard braking while following the lead car

briskspirit, 2020-11-20:
**Describe the bug**

Was following the lead car on a highway traffic jam, that car was going without lights so might be a reason. Braking was really hard when he stopped, almost hit him ) I had a feeling that C2 don't see it at all.

**How to reproduce or log data**

Don't know. Get a white lead car without lights? (Later when he switched them on my car was braking fine)

**Expected behavior**

More soft braking

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: f66032c2b5aa18ac
 - Route: f66032c2b5aa18ac
 - Timestamp: 2020-11-19--17-35-53--23
 - Version: 0.8,0 (already updated to latest commit, can't get previous hash)
 - Car make/model: Toyota Corolla 2021

**Additional context**

@Torq_boi said that it is not a model bug, but old known problem with no time to brake as lead car accelerated and braked fast. (So could INDI tuning fix that problem?)

[image]

pd0wm, 2020-11-23:
Looks like the toyota ACC doesn't correctly responds to our actuator commands causing the controller to wind up. This happens in some other situations too. Would be nice if we had a real way of detecting this so we can prevent controller windup.

adeebshihadeh, 2022-06-12:
Duplicate of #2731

closed 2022-06-12 by adeebshihadeh