commaai/openpilot #2596: [0.8] Auto lane change crosses lane edge in directionlhofQWERT8EZ123456789-=====================================================================================

abdulrahman004, 2020-11-21:
Stopped car detection has been working pretty well on 0.8 Came across this instance where I had to apply brakes because lead car wasnt detected until then. May be because the lead car was up hill ?

Expected behavior
Model should have detected stop light at signal

Device: FreEon
Dongle ID: 59f457ce47259dd2
Route: 59f457ce47259dd2|2020-11-19--02-55-03--5
Timestamp: 19:00:11
Version: a359d8a
Car make/model: Toyota Sienna 2018
Additional context

gif of the report
https://imgur.com/a/e6BpEez

C0mpl3t3N00b, 2020-11-24:
I experience this too on my 2020 Rav4 Hybrid LE with v0.8

HaraldSchafer, 2020-11-26:
In 0.8 the model now understands how high the camera is above the road. Previously it did not and it always assumed it was 1.22m above the road.

This meant that for tall cars it would think lanelines were closer together than they actually were and lane changes were smaller movements then they really were. Since they looked smaller in the image and the model did not understand the concept of height. 

I suspect that tall cars were previously tuned to compensate for this and that now that it is fixed tall cars ping pong and overshoot lane changes on that tune.

I still need to confirm this definitively, but if this is true tall cars may need a retune for 0.8. Will post an update here if I find out more.

abdulrahman004, 2020-11-27:
Harald: Let me know if you need more data on specific type of roads or specific situations. Happy to test out and report back. Also, if there is specific tuning/params you want me to try; let me know. I will take my laptop along with me and change them multiple times for the same route

ErikRNissen, 2020-11-28:
Very similar issue with 2018 Volt (Not a tall car)
Dongle ID - (ae3ed0eb20960a20)
Drive - https://my.comma.ai/ae3ed0eb20960a20/1606584122502/1606584849065
  
Time stamp - 3 events
 1 - 9:25:52 overshot lane change on straight section of road
 2 - 9:26:02 overshot into 3rd lane going to the left on a straight section of interstate
 3 - 9:26:14  lane change to the right on a straight road - so slow I had to take the wheel to complete lane change

y-a-n-w-a-n-g, 2020-11-29:
0.8 release2-staging on Thanksgiving day with a Corolla Hatchback 2020

I had ALC try to change across multiple lanes on curvy roads two separate times.
Only managed to find one clip -  [https://imgur.com/8FO8RuB](https://imgur.com/8FO8RuB), but the same thing happened for both of them, where the car starts going into the second lane to the left and I have to pull it back.

Memberoffoxhound, 2020-11-30:
2018 Camry SE here. Happened in Master CI as well as release now. I'll be sure to note and post the route ID. It's intermittent for me.

C0mpl3t3N00b, 2020-12-04:
On release2 and it happened again that I could remember to to look up:

bdad50b8a1f9fa8e|2020-12-03--09-34-21--8      (09:42:50)

If a car was in the other lane, contact would have occured.

gilton123, 2020-12-05:
also on my 2019 corolla hatchback

abdulrahman004, 2020-12-28:
@HaraldSchafer I tested this again on the new release version 0.8.1 and it seems this doesnt happen anymore on the few tests I did. I am not sure if there was something specifically done to address this issue in the new version or it was a side affect of other changes. Nevertheless, I am inclined to close this issue unless somebody is still experiencing this

@C0mpl3t3N00b @gilton123 @y-a-n-w-a-n-g Are you folks still seeing the issue in 0.8.1 ?

HaraldSchafer, 2020-12-28:
Several things were fixed in 0.8.1, including a few things that affect laneless path planning, which includes lane changes. 

But not sure if this specifically was fixed, feel free to close it if you feel it is no longer relevant.

RCoaster, 2021-01-01:
Actually, I had no issue with 0.8.0 on my 2019 Subaru Ascent, but with 0.8.1, the issue was there. There would be times where it would do it successfully but come close to the line, and other times it would cross the lane enough to trigger a lane departure warning.

y-a-n-w-a-n-g, 2021-01-02:
On master-ci 0.8.2 (not sure which version) it happened twice for me.  Think  it was around the beginning of Christmas week.

dibble9012, 2021-01-03:
I have had it multiple times on 0.8.1 on my cross country trip this past week.

HaraldSchafer, 2021-03-03:
Similar problems https://github.com/commaai/openpilot/issues/19865

HaraldSchafer, 2021-03-31:
Improved in 0.8.3

closed 2021-03-31 by HaraldSchafer