commaai/openpilot #2616: eigen,capnp,zmq  compile bug for mac

gucasbrg, 2020-11-24:
'eigen,capnp,zmq' configuration  can be finded in In earlier Makefile versions,

```
EIGEN_FLAGS = -I$(PHONELIBS)/eigen
ifeq ($(UNAME_S),Darwin)
  VT_LDFLAGS += $(PHONELIBS)/capnp-c/mac/lib/libcapnp_c.a \
                 $(PHONELIBS)/zmq/mac/lib/libzmq.a \
                -framework OpenCL
```
but upgrade to scons, in mac it seems to incorrectly configured
for command 
```
cd cereal &&  scons -c && scons -j$(nproc)
```

wrong info output

```
	scons: Reading SConscript files ...
	scons: done reading SConscript files.
	scons: Cleaning targets ...
	scons: done cleaning targets.
	scons: Reading SConscript files ...
	scons: done reading SConscript files.
	scons: Building targets ...
	mkdir -p gen/c/include && touch gen/c/include/c++.capnp.h gen/c/include/java.capnp.h
	capnpc --src-prefix=/Users/one/openpilot/cereal car.capnp log.capnp -o c++:gen/cpp/
	python3 /Users/one/openpilot/cereal/services.py > services.h
	clang++ -o messaging/msgq.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include messaging/msgq.cc
	clang++ -o gen/cpp/car.capnp.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include gen/cpp/car.capnp.c++
	clang++ -o messaging/socketmaster.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include messaging/socketmaster.cc
	clang++ -o messaging/impl_zmq.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include messaging/impl_zmq.cc
	In file included from gen/cpp/car.capnp.c++:4:
	gen/cpp/car.capnp.h:7:10: fatal error: 'capnp/generated-header-support.h' file not found
	#include <capnp/generated-header-support.h>
	         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	clang++ -o messaging/bridge.o -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -I. -I/usr/lib/include messaging/bridge.cc
	clang++ -o messaging/impl_msgq.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include messaging/impl_msgq.cc
	clang++ -o messaging/messaging.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include messaging/messaging.cc
	1 error generated.
	scons: *** [gen/cpp/car.capnp.os] Error 1
	clang++ -o gen/cpp/log.capnp.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -fPIC -I. -I/usr/lib/include gen/cpp/log.capnp.c++
	In file included from gen/cpp/log.capnp.c++:4:
	gen/cpp/log.capnp.h:7:10: fatal error: 'capnp/generated-header-support.h' file not found
	#include <capnp/generated-header-support.h>
	         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	1 error generated.
	scons: *** [gen/cpp/log.capnp.os] Error 1
	In file included from messaging/socketmaster.cc:3:
	messaging/messaging.hpp:6:10: fatal error: 'capnp/serialize.h' file not found
	#include <capnp/serialize.h>
	         ^~~~~~~~~~~~~~~~~~~
	In file included from messaging/messaging.cc:1:
	messaging/messaging.hpp:6:10: fatal error: 'capnp/serialize.h' file not found
	#include <capnp/serialize.h>
	         ^~~~~~~~~~~~~~~~~~~
	messaging/impl_zmq.cc:7:10: fatal error: 'zmq.h' file not found
	#include <zmq.h>
	         ^~~~~~~
	1 error generated.
	scons: *** [messaging/socketmaster.os] Error 1
	In file included from messaging/bridge.cc:11:
	In file included from messaging/impl_msgq.hpp:2:
	messaging/messaging.hpp:6:10: fatal error: 'capnp/serialize.h' file not found
	#include <capnp/serialize.h>
	         ^~~~~~~~~~~~~~~~~~~
	1 error generated.
	scons: *** [messaging/messaging.os] Error 1
	In file included from messaging/impl_msgq.cc:9:
	In file included from messaging/impl_msgq.hpp:2:
	messaging/messaging.hpp:6:10: fatal error: 'capnp/serialize.h' file not found
	#include <capnp/serialize.h>
	         ^~~~~~~~~~~~~~~~~~~
	1 error generated.
	scons: *** [messaging/impl_zmq.os] Error 1
	1 error generated.
	scons: *** [messaging/bridge.o] Error 1
	1 error generated.
	scons: *** [messaging/impl_msgq.os] Error 1
	scons: building terminated because of errors.

```

pd0wm, 2020-11-24:
Does it work if you call `scons cereal` from `openpilot` instead of from the cereal?

gucasbrg, 2020-11-24:
> Does it work if you call `scons cereal` from `openpilot` instead of from the cereal?

Seems to be the same problem
```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
clang++ -o cereal/gen/cpp/car.capnp.os -c -std=c++1z -DGL_SILENCE_DEPRECATION -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -DGL_SILENCE_DEPRECATION -fPIC -Iexternal/tensorflow/include -I. -Iselfdrive -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iselfdrive/boardd -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal/messaging -Icereal -Iopendbc/can -Iphonelibs/json11 cereal/gen/cpp/car.capnp.c++
In file included from cereal/gen/cpp/car.capnp.c++:4:
cereal/gen/cpp/car.capnp.h:7:10: fatal error: 'capnp/generated-header-support.h' file not found
#include <capnp/generated-header-support.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
scons: *** [cereal/gen/cpp/car.capnp.os] Error 1
scons: building terminated because of errors.
```

gucasbrg, 2020-11-24:
@pd0wm in SConstruct file 
modify 
```
   cpppath = [
     "#external/tensorflow/include",
+    "/usr/local/include",
   ]
 

```
and can work prefect

closed 2020-11-24 by gucasbrg

pd0wm, 2020-11-24:
How did you install capnproto? We run a macos build in CI that still works, so it should work without changes.

This is the setup script we run in CI: https://github.com/commaai/openpilot/blob/master/tools/mac_setup.sh