commaai/openpilot #2624: OpenPilot Fails to Stop for Stationary Lead Car

C0mpl3t3N00b, 2020-11-25:
**Describe the bug**

**How to reproduce or log data**

Approaching a red light and following a lead car with OpenPilot engaged, my car blared a bunch of collision warnings and I had to disengage and press my brake pedal to avoid hitting the lead car. 

**Expected behavior**

OpenPilot should stop itself if a stopped car is ahead.

**Device/Version information (please complete the following information):**
 - Device: Comma two
 - Dongle ID: bdad50b8a1f9fa8e
 - Route: bdad50b8a1f9fa8e|2020-11-23--13-17-12--12
 - Timestamp: bdad50b8a1f9fa8e|2020-11-23--13-17-12--12
 - Version: master-ci
 - Car make/model: 2020 Rav4 Hybrid LE

**Additional context**

Previous problem was noted here: https://github.com/commaai/openpilot/issues/2484 but was closed since it was not a "release" build. Same issue still occurs on master-ci.

adeebshihadeh, 2020-11-25:
Duplicate of #2574.

closed 2020-11-25 by adeebshihadeh

C0mpl3t3N00b, 2020-11-25:
> Duplicate of #2574.

I added this in case Device made a difference; other post was for a non official Comma device (FrEON).