commaai/openpilot #2627: Long control on Nidec

hondacm450e, 2020-11-25:
Long control leads to more disengagement 

Honda Civic 2016 Nidec 

Openpilot has come along way I’m lateral control. As long control leaves something to the desire. This is an argument to shorten the following distance please read. 

I’ve noticed that long control loses the lead car because the distance so far that I can’t track around curves as well and as such in losing and regaining vision of the lead car loan control his poor performance. This leads to further disengagement’s.
 secondlyI noticed that I have to disengage whenever a car pulls in front of me Because open pilot falls back so quickly that other cars make aggressive decisions around my car. Even if I disengage and reset it to see if following distance openpilot again slows down creating a pattern of erratic long control. In this instance you are also increasing disengagements. 

Please consider shortening follow distance Nidec Hondas to reduce these disengagements. Or at least please try a kegmans branch on 1 and 2 bar follow distance on your civic. I have consistently had less disengagement using those following distances and felt the overall drive is safer then your base follow distance, as the the car is more interpretable to other drivers on the road and more predictable to the traffic pattern.

pd0wm, 2020-11-26:
We won't be changing the following distance until there is end-to-end longitudinal control.

doktorsleepelss, 2020-11-26:
> Because open pilot falls back so quickly that other cars make aggressive decisions around my car.

Get yourself a comma pedal.  A lot of that has to do with slow acceleration of the nidec system that openpilot can't change without a pedal.

closed 2020-11-30 by pd0wm