commaai/openpilot #2632: Subaru Crosstrek 2018 keeps saying "Car Unrecognized"

DannyHibbard, 2020-11-26:
**Describe the bug**

Whenever I start the car up, the screen just say's "Car Unrecognized" and goes into Dashcam mode. It's worked twice, where I went into calibration mode and set it up the first time, and the car drove me the second time. Every other time it's just come up with this error.

**How to reproduce or log data**

Start up the car

**Expected behavior**

I'd expect the comma to behave as it should, taking over the controls when cruise control is activated.

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [44f74c0e55ca7c27]
 - Route: [44f74c0e55ca7c27 2020-11-26--16-46-32]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [0.7.6)]
 - Car make/model: [Subaru Crosstrek 2018 (UK)]

**Additional context**

pd0wm, 2020-11-27:
I see you also ran a fork which does recognize your car. If you submit a PR with the fingerprint for your car we can add support.

closed 2020-12-08 by adeebshihadeh

DannyHibbard, 2021-01-25:
Hi, I'm so sorry for the delay in getting back to you @pd0wm 

How would I go about submitting a PF with the fingerprint? Would this mean I'd just use the official open pilot and it would work? 

Unfortunately, for some reason, my vehicle is no longer recognised with he branch I was using, and the MartinL branch is also giving me errors - "White and grey panda are usupported. Upgrade to comma two or black panda"

Thank you in advance, and sorry again for the delay!

amd006, 2021-05-15:
I'm having the same problem with my Subaru Crosstrek 2019. Is it because my phone is not pairing with the dash cam?