commaai/openpilot #2633: Restarting the comma2 with vehicle running causes dash error which only clears up after vehicle restart

xps-genesis, 2020-11-26:
**Describe the bug**

On the Hyundai Genesis 2015-16, when the comma2 starts up the LKAS fault occurs followed by SCC not present then the AEB faults permanently. The AEB fault needs a vehicle restart to clear. 

Other observed behavior:
* The vehicle is fine if the comma to is just shutdown and it is also fine when the device is disconnected from USB(after shutdown). The issue is only present when the comma2 starts and the vehicle is already running.
* During the comma2 restart, I assume there a brief moment when the relay cuts out which is not allowing the cam messages to pass through which trickles down to AEB faulting permanently. 

**How to reproduce or log data**

Restart the comma 2 which the car running

**Expected behavior**

No faults on the dash

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: []
 - Route: [N/A]
 - Timestamp: [N/A]
 - Version: [0.80 master-ci ( not related to this version- the issue was always present)]
 - Car make/model: [Hyundai Genesis 2015-16]

**Additional context**

I want to know if there is any obvious fix that can get rid of this behavior

pd0wm, 2020-11-27:
I don't think there is anything we can do about this. Honda Nidec and Toyota's with DSU unplugged also fault permanently when you restart the C2.

xps-genesis, 2020-11-27:
ok.. may I know what happens during the c2 startup that might be causing this issue?

pd0wm, 2020-11-27:
When the panda loses the heartbeat signal the safety mode resets to something like no output. The forwarding stops which is probably why the car faults. But depends on your harness setup of course.

closed 2020-11-30 by pd0wm