commaai/openpilot #2649: Once engaged, automatically steers to the right without stopping

gnuel3, 2020-11-29:
Once I upgraded to 0.8 comma two on my 2018 Toyota Prius prime, once I engage openpilot, it just steers to the right no matter what. Very scary.  I tried restarting many times and same issue Persists. This has Happened since 3 days ago. I did not change the mount or anything. My dongle id is 

A91b34765fc70109

Before the update, it worked beautifully and amazing

pd0wm, 2020-11-30:
It looks like your GPS is not working, which is causing these issues. Are you not getting an alert about that every drive? Please contact support@comma.ai to get this hardware problem fixed.

closed 2020-11-30 by pd0wm

gnuel3, 2021-01-05:
> It looks like your GPS is not working, which is causing these issues. Are you not getting an alert about that every drive? Please contact  support@comma.ai (mailto:support@comma.ai)  to get this hardware problem fixed.

Hi William,
  

  
My new comma one suddenly stoped working. It just shows the comma logo and keeps making resetting noises. I tried unplugging it and plugging back in to no avail.
  

  
  
  

  
 Would you be able to check what’s wrong?   
  

  
Thank you,
  
Albert Leung