commaai/openpilot #2650: OP did not slow down behind a stopped car. Was it late in detecting?

wongjunjin, 2020-11-29:
**Describe the bug**
My car was cruising and was met with a traffic stop. However, my car did not slow down until I intervene. There were no other cars beside me. However, maybe if you can take a look and see what was wrong with the model.

**How to reproduce or log data**
Unsure.

**Expected behavior**
Car should have slowed down earlier.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 7bc774bc8e17fd8a
 - Route: 7bc774bc8e17fd8a|2020-11-29--08-16-51
 - Timestamp: 29 Nov 2020 @ 16:24:47
 - Version: 0.8.0
 - Car make/model: Toyota Corolla 2019

scottrcarlson, 2020-11-29:
Describe the bug
I also observed similar behavior, on Friday afternoon. I expected car to stop behind lead car a a traffic light on a two-lane highway. We were moving quick, and I disengaged for safety. I would have expected to begin deceleration, based on the speed we were traveling.

How to reproduce or log data
Have only seen once. No reproducible instructions at this time.

Expected behavior
Expected deceleration to begin sooner. 

Device/Version information (please complete the following information):

Device: comma two
Dongle ID: d7b585d5aab7a139
Route: d7b585d5aab7a139|2020-11-27--14-26-13--10
Timestamp: 27 Nov 2020 @ 14:36:15
Version: 0.8.0
Car make/model: Toyota Corolla 2019 hatchback

pd0wm, 2020-11-30:
@wongjunjin openpilot had a little trouble because the stationary car was hidden behind another car that eventually turned left. You took over right as at begun braking.

@scottrcarlson the full logs are not uploaded yet for your drive, so can't look into it. If you connect to wifi they will upload.

wongjunjin, 2020-11-30:
> @wongjunjin openpilot had a little trouble because the stationary car was hidden behind another car that eventually turned left. You took over right as at begun braking.
> 👍🏻

scottrcarlson, 2020-12-01:
> @wongjunjin openpilot had a little trouble because the stationary car was hidden behind another car that eventually turned left. You took over right as at begun braking.
>
> @scottrcarlson the full [...]

Woops. I just corrected that, it's connected and should be uploaded shortly.

-Scott

scottrcarlson, 2020-12-08:
I'm under the impression that coming in hot on a stationary vehicle is a well known behavior. I assume OP would have initiated braking shortly after I disengaged, Toyota's Radar hadn't detected the object yet. 

Reading the forums and driving a bit more, I don't believe this is an unknown behavior. 

Does that sound about right?

Thanks!

pd0wm, 2020-12-10:
Correctly detection stationary objects is hard. Radar has a lot of false positives for stationary objects, so confirmation with vision is always necessary. This is also mentioned in the readme: https://github.com/commaai/openpilot#limitations-of-openpilot-acc-and-fcw

closed 2021-06-16 by pd0wm