commaai/openpilot #2651: Received "Communication Issue Between Processes" while driving

wongjunjin, 2020-11-29:
**Describe the bug**
I received the "Communication Issue Between Processes" while driving. I am not sure what triggered it but it happened for the first time since I had comma2. And it happened a few times in today's drive. 

**How to reproduce or log data**
Unsure

**Expected behavior**
What should I have done?

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 7bc774bc8e17fd8a
 - Route: 7bc774bc8e17fd8a|2020-11-29--08-16-51
 - Timestamp: Nov 29 @ 16:26:40 & 16:27:43 
 - Version: v0.8.0
- Car make/model: Toyota Corolla 2019

nickgeurnop, 2020-11-30:
**Describe the bug**

I am receiving the same kind of errors as described in the original bug report. The car will throw "Communication Issue Between Processes" while driving. I had this happen 4 times in one drive. On this particular drive, it went away and I was able to complete over 150 miles without the issue coming back. 

**How to reproduce or log data**

I drove 160 miles error free until I stopped for gas/restroom. I turned off the car, went to the restroom for 5 minutes. I turned the car back on to pull to the gas pump. Turned car off. Filled up with gas for <5 minutes. Turned car back on. 

**Expected behavior**

The system can communicate with the car and operate as intended without the need for a planned disengagement.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: 1615388ccbe3edc9
 - Route: 1615388ccbe3edc9|2020-11-29--15-05-41
 - Timestamp: Nov 29 @ 15:11:12 & 15:15:32 & 15:21:12 & 16:00:23
 - Version: release2 v0.8.0
 - Car make/model: 2019 Toyota Corolla Hatchback XSE (TSS2.0)

**Additional context**

I have not experienced this behavior prior to release2 0.8.0

The openpilot Icon on UI would flicker on and off, turning the "main cruise" off and on made show up as intended

No restarts. no shutting the car off, no reconnecting connectors. It just "went away" mid-drive.

pd0wm, 2020-11-30:
Confirmed both issues are the same as https://github.com/commaai/openpilot/issues/2636.

closed 2020-11-30 by pd0wm