commaai/openpilot #2654: Updates should only be installed when the user consents to the update

NSExceptional, 2020-11-30:
**Describe the bug**

A pending update was installed without my consent.

**How to reproduce or log data**

In my case, the update installed itself while I was on the road after my Comma got a CAN connectivity issue and I had to unplug and replug my device, rebooting it. All of my changes were discarded which can be dangerous and unexpected in a situation like that.

**Expected behavior**

Updates should only occur ***WHEN THE USER CHOOSES TO UPDATE*** and there should be a setting to disable automatic update checking entirely imo.

adeebshihadeh, 2020-11-30:
This is the intended functionality and not a bug. When the car is turned off, your device would have shown an alert with the release notes and new version, but you rebooted before that. There is a parameter called `DisableUpdates` which disables automatic updates, but it's intentionally not exposed in the UI.

closed 2020-11-30 by adeebshihadeh

NSExceptional, 2020-11-30:
Could you elaborate for me how it was installed with out me clicking a button? Why does rebooting install it? I've been seeing the prompt for days and rejected it every time.

Thank you, where can I find documentation on this parameter?

adeebshihadeh, 2020-11-30:
On boot, we check if there's an update ready and perform the update if so. If you simply dismiss the update alert, it won't update. Here's the param for disabling updates - https://github.com/commaai/openpilot/pull/1807.

NSExceptional, 2020-12-13:
@adeebshihadeh So I've done this, but now I'm getting a warning telling me OpenPilot will stop working until it can check for updates? What's up with that?