commaai/openpilot #2656: I think my temperture indicator is broken reading 30 degrees C (86 F) the fan is running constantly

Kay585, 2020-11-30:
It is cloudy and 50 outside but the temperature indicator is reading 30 degrees (86 F).  This is causing the fan to run constantly.  Anything to fix this problem?    Also if a upgrade to Comma 2 - is it a drop in replacement? 

- Device: Comma ai  
- Route:  Toyota RAV 4 2017

pd0wm, 2020-12-01:
What is your dongle id?

The CPU is getting quite a bit hotter than 30 C, so that's why the fan has to run.

adeebshihadeh, 2020-12-03:
Closing due to inactivity. Ambient temperature doesn't reflect internal temperature of the components of the device.

closed 2020-12-03 by adeebshihadeh

Kay585, 2020-12-03:
@adeebshihadeh  -- Please consider reopening.  The unit when first turned on has this high temperature so I dought the internal temp is that high when first starting.  openpolot V0.8.0-release Git Brand release 2 Git Revision 1c71239216a0.  Do you need more info?

Kay585, 2020-12-03:
@pd0wm  -- Is this the Dongle ID GIT VERSION 1c71239216a0

pd0wm, 2020-12-03:
dongle id is in settings, device, dongle id