commaai/openpilot #2660: Going to device settings causes BaseUI to stop

xps-genesis, 2020-11-30:
**Describe the bug**

go to setting, devices. The base ui stops. settings is not available anymore. off-road display is blank.

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [Comma two]
 - Dongle ID: [6fe86b4e410e4c37]
 - Route: [NA]
 - Timestamp: [NA]
 - Version: [master-ci]
 - Car make/model: [Hyundai Genesis 2016]

**Additional context**

[image]

adeebshihadeh, 2020-11-30:
Is this with stock openpilot (including the APK)? If so, include the commit in your bug report.

gugupilot, 2020-11-30:
> 
> 
> Is this with stock openpilot (including the APK)? If so, include the commit in your bug report.

yes, just copied the master-ci. updated with details

adeebshihadeh, 2020-12-01:
I can't reproduce on latest master. All of your latest routes are from a fork. Feel free to reopen if you can reproduce reliably on master or master-ci, not a fork rebased on master.

adeebshihadeh, 2020-12-08:
This should be fixed as of 9803d3fcdc9ceef32ec56b79dbe7194b164393e6.

closed 2020-12-08 by adeebshihadeh