commaai/openpilot #2671: openpilot accelerates behind a stopped car

doktorsleepelss, 2020-12-01:
**Describe the bug**

After my car almost came to a complete stop behind another stopped car, openpilot decided to unnecessarily accelerate causing me to manually brake hard. 

**How to reproduce or log data**

Was speed was close to 50 mph following another car, and the a red light caused the cars to stop in front of me. My car also braked, but the braking started relatively late, which cause the first forward collision warning.  Ultimately, my car did seem to brake just in time for a smooth stop, but right before reaching 0mph, it randomly decided to accelerate with no movement from the car in front of me. This caused another forward collision warning before I manually braked.

One thing that maybe relavent is that because of the late braking, I did feel as if my car stopped (or got near stopping) closer behind the car in front of me than it probably would have normally. 

**Expected behavior**

My car should have come to a complete stop, and only accelerate once  the lead car started to move again.

**Device/Version information (please complete the following information):**
 - Device: Eon Gold with black panda
 - Dongle ID: 23b69975490100e4|
 - Route: 23b69975490100e4|2020-11-30--12-14-05--5
 - Timestamp: 340 seconds into the route or end of the segment
 - Version: 0.8 release2-staging fork with a few pedal related changes https://github.com/doktorsleepelss/openpilot/commits/release2-staging
 - Car make/model: 2016 honda civic sedan

https://imgur.com/148Pjx8

https://my.comma.ai/cabana/?route=23b69975490100e4%7C2020-11-30--12-14-05&exp=1638398245&sig=Kef41qsWemGBZHUfY0DLyO6vIo5TmQs8PAPchMpklDQ%3D&max=27&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F23b69975490100e4%2Fca7efd06a33d9236b458b35e0b77aca3_2020-11-30--12-14-05

pd0wm, 2020-12-03:
This looks to be caused by your tuning for the pedal. The tuning can probably improved in general, but there is a clear bug in how you cap the maximum gas value here: https://github.com/doktorsleepelss/openpilot/commit/c54202c44d548d4b479ccc69fa26a79b288e575e#diff-8f6df5d986fd4f23c0c08c3737498ef5b4a270f3919a76cd8dd9678401f8c8caR426

If you cap the actuator value you have to make sure to also cap the integral of the PID at the same value. Now the integral just goes to 1 because it tries to accelerate but you're limiting the acceleration downstream. It then takes quite a bit of time to wind down again when reach the desired speed.

In the segment you shared the integral reached 1 when pulling away at the traffic light, and was still pretty high after the braking maneuver. When it tried to move forward it overshot due to the high integral.

doktorsleepelss, 2020-12-03:
Thanks for the feedback. I kind of suspected it would end up being pedal related as I mentioned in the discord. But I'm glad it wasn't model related.

closed 2020-12-11 by pd0wm