commaai/openpilot #2677: C2 drains/kills Prius car battery (0.5-0.8) - likely not software related

zorrobyte, 2020-12-02:
**Describe the bug**

Running 0.8 and my Prius Prime's 12V battery is dead again :D I don't drive the Prius but once a week or so. Considering the C2's blue light *always* flashes, it doesn't seem that software is going to solve the battery drain issue. I've had 10+ dead batteries since having OP in my car; got stranded while out camping once, too. It's quite annoying and especially concerning now that it's winter.

**How to reproduce or log data**

Leave C2 installed in a vehicle with a smaller 12V battery, like a Prius/Prius Prime. Wait a week = dead battery.

**Expected behavior**

Not being stranded.

**Device/Version information (please complete the following information):**
Issue has been present for me since 0.5 across FrEON and C2

**Additional context**

The blue light flashes/is on the rear of the C2 even when the unit is turned off. Parasitic power drain is too high for those who don't drive their vehicles everyday and have smaller 12V batteries.

zorrobyte, 2020-12-02:
This module resolves the issue, but is a 3rd party device:
https://www.amazon.com/Priority-12-Volt-Automatic-Battery-Protector/dp/B071GVCHHC/ref=sr_1_5?dchild=1&keywords=priority+start&qid=1606925819&sr=8-5

ErichMoraga, 2020-12-02:
Last time I measured the C2's drain in sleep mode, it was under 50ma, though I can break out my logger and check again tonight.  Don't forget our cars perform a semi-problematic evap. leak test at 5 hours (as seen in the manual) which causes a measurable power drain too.  Once a Prius Prime gets a low spike in a certain range, the ECUs freak out, and create huge power drain causing other systems to nosedive (you'll hear a clicking in the motor/generator, for example).

ErichMoraga, 2020-12-02:
I did some preliminary testing, bearing in mind last time I did this was in June on much older code.  Here's what it looked like when I would park my car back then (head/tail of captured output)...

```
Volts  mA
13.884 347.890
13.908 330.293
13.852 271.195
12.328 141.000
12.412 208.695
12.488 238.098
12.488 135.293
12.464 165.293
12.480 156.598
12.504 119.500
...
12.280 42.195
12.280 36.000
12.280 41.293
12.280 41.793
12.284 39.695
12.280 42.695
12.284 41.098
12.280 41.902
12.284 35.500
12.280 39.402
```

My C2 has been on 0.8.0 for several days, though I haven't driven for a couple days, and did a preliminary test with my car staying off.  First was with the C2 powered on...
```
11.920 141.500
11.876 160.793
```

Then I powered off remotely via `reboot -p`, so the C2 was asleep...
```
11.872 176.988
11.872 183.390
```

Shocked by the #s, came back to the C2, powered it back on, then off via the power button and UI, and confirmed asleep...
```
11.880 153.293
11.764 81.793
11.868 173.988
```

zorrobyte, 2020-12-02:
Just to note, Prius 12v battery is very small. At 46Ah (total capacity), the battery will be drained to 0% in 11 days~ at 0.173 amps
[image]

Considering that lead acid/AGM batteries should never be discharged past 50% (and that's for "deep cycle" rated batteries, not starting batteries, like in our Priuses), it becomes:
[image]
5.5 days

Even at an "optimal" draw of 0.042A, the reserve capacity to 50% depth of discharge is only 22 days:
[image]
Considering a brand new battery.

Just one time of a 100% discharge can reduce the total capacity of a lead-acid starting battery by 50%, then the usable capacity **halves**, leading to a dead battery cycle where a costly replacement is a guarantee.

vanillagorillaa, 2020-12-02:
The comma pedal only draws 5v from the cars pedal connector, not the comma power.

ErichMoraga, 2020-12-03:
Right, I told Zorro about that on the phone earlier (fake eth. spec 1 vs. 2).  Anyway, I did a proper drive to charge up my car earlier, and have my tool running for the next 6 hours w/ the 1st 3 hours having my C2 on, and the last w/ it off in sleep mode.  Will post results once done.

codymhorton, 2020-12-03:
Just wanted to +1 this. I just had my battery die after 3-4 days of not using my car. Went to start it and had to jump it off.

@ErichMoraga I actually asked you about this a little while back and you linked [this](https://github.com/ErichMoraga/openpilot/commit/6717ce7db6cd930cb53fc1dadcc49a479ee192eb) could this fix the issue or is it unrelated.

I have a 2017 Prius with dsu unplugged, C2 of course.

For now I'm back to unplugging it after each use...

ErichMoraga, 2020-12-03:
See above where I posted the previous normal power draw older code had (C2 on vs. off w/ Uno's BP asleep).  That's where my commit (which you cited) clearly shined.  Now look at the current metrics, Cody, and you'll see what the fuss is about.  Anyway, my latest metrics will be ready in a few more hours.

ErichMoraga, 2020-12-03:
4.5 hours into the 6 hour test, it's very clear there's a problem.  Phone has been off, sleeping for 1.5 hours, and I'll let this sample of the data speak (sampled every minute)...
```
...
12.328 166.695
12.352 72.500
12.328 170.293
12.328 61.500
12.324 168.793
12.328 170.293
12.324 163.000
12.324 164.695
12.324 165.195
12.320 170.890
12.348 73.695
12.320 169.793
12.348 76.098
12.320 170.598
12.320 170.500
12.352 35.195
12.320 170.988
12.320 169.293
12.340 76.195
12.344 117.500
12.316 170.500
12.320 168.890
12.316 171.390
12.316 170.598
12.316 170.098
12.324 134.390
12.316 162.390
12.320 155.390
12.324 164.000
12.312 169.000
12.312 144.500
12.312 176.488
12.332 75.793
12.308 176.390
12.308 177.390
12.336 82.098
12.312 160.293
12.308 166.793
12.308 170.195
12.308 172.890
12.308 171.793
12.308 170.988
12.308 169.098
12.308 171.598
12.332 76.098
12.300 170.598
12.308 170.695
12.304 171.598
12.304 166.390
12.304 166.000
12.304 169.195
12.304 171.293
12.300 157.195
12.308 168.195
12.304 155.000
12.300 167.000
12.308 145.793
12.300 168.195
12.300 177.390
12.300 171.988
12.296 179.293
12.300 146.793
12.296 169.195
```

ErichMoraga, 2020-12-03:
Chart...
https://drive.google.com/file/d/1sfs6oYdlCwQKirtqe3DUEzP7hh2lAFxS/view?usp=sharing

Data used to create chart...
https://drive.google.com/file/d/1NiP-Oqy7J0fcrnaz0bKKirj0xagBPEWG/view?usp=sharing, 

Also posted in private channel in Discord...
https://discord.com/channels/469524606043160576/532205411709747212/783953369910738944

robbederks, 2020-12-03:
It started happening with this PR: https://github.com/commaai/panda/pull/611, which doesn't turn off the power to the phone when it gets turned off, to retain the functionality of the power button.
Turns out that we also had a subtle bug in the bootkick circuitry/code which makes some devices not turn off completely, causing the elevated power draw of up to 1.5W.
Fixed in https://github.com/commaai/panda/pull/614

ErichMoraga, 2020-12-03:
Nice catch, thanks!

papagunit, 2020-12-03:
Glad this was acknowledged, I thought it was due to my battery going out on my '16, however unplugging panda would stop the drain and it wasn't dying after 48 hours if non use.

zorrobyte, 2020-12-04:
Whohoo, awesome work!

Just wanted to say that there are still edge cases where OP software can crash leading to a black backlit device state that will rapidly drain the battery; something that happened when I was out camping (was stranded 8 hours later), and another time the last week. Still, some sort of low voltage disconnect seems critical for the next/consumer release of hardware as a sanity check as getting stranded is bad news bears.

Closing!

closed 2020-12-04 by zorrobyte