commaai/openpilot #2688: Manager failed to start

adam-nelson, 2020-12-03:
**Describe the bug**

Recent update has caused an error startup? I am on master-ci 0.8.1

**How to reproduce or log data**

Update comma two to latest revision

**Expected behavior**

No errors on startup

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [18e8ac91b805db11]
 - Route: [unknown]
 - Timestamp: [3am local time aest]
 - Version: [master-ci 0.8.1]
 - Car make/model: [Toyota corolla accent sport hybrid 2019]

**Additional context**

[image][image]

pd0wm, 2020-12-03:
Fixed by https://github.com/commaai/openpilot/commit/d5a806e564afcc1b4de0b934a6073e4695cc7f96

closed 2020-12-03 by pd0wm

adam-nelson, 2020-12-03:
Dam typos